df8oe/UHSDR #1426: please set the low border of DSP FIR NUM TAP higher

DG9BFC, 2018-02-07:
when you "accidently" set the FIR NUM TAP to zero "0" and then activate notch you crash the cpu (no wonder it can not work with zero filter taps)
so please set that to 16 as minimum border

DD4WH, 2018-02-07:
thank you Sigi!
done in the next build!
73 Frank

DG9BFC, 2018-02-09:
ok min end is set to 16 (setting to zero impossible) issue can be closed
but what about max end?? 32 is not that much and the mchf cpu can do better

df8oe, 2018-02-09:
No. F4 **cannot do better**. We have had to reduce this because it does not work under all circumstances.

DG9BFC, 2018-02-09:
yes i know the Cpu is on the max load edge when you engage all gimmicks ... but why then lower the values so much? ... only when you have all things engaged (nr+notch+nb+ wider filter + high zoom ratio plus ??? whatever) then cpu puzzles the audio buffers somewhat ... with long press of menue these buffers are again "brought in the line" and then the sound is clear again (just tested with 94 version here) ... it is not only problem of cpu load but also that the buffers can become somewhat mixed with a high cpu load
lowering the values too much will lower performance of the mchf ... mchf is not a ferrari ... but we are driving a "golf with enganged hand brake"  if we lower the values more as needed

DD4WH, 2018-02-09:
Hi Sigi,
the bug (crash when no. of taps set to zero) is fixed already.
The reason for fixing the no. of taps to 32 is not the processor power.
It is the observation, that the buffer length of the NR autonotch LMS filter has to be an integer multiple of the no. of taps. (And the observation that the no. of taps has virtually NO influence on the quality of the autonotch from my experience).
So in the next version I will eliminate the menu options "notch no. taps" and "notch buff len" from the menu and fix these values to 128 buff len and 64 no of taps. And I will (experimentally!) increase the max, convergence rate that is user adjustable. 
73 Frank

DG9BFC, 2018-02-09:
hmmm length has to be an integer multiple of no. of taps?? i did not know that ... how can i reproduce here?
ok ... the issue with number set to zero IS SOLVED so i will close the issue here .. the rest can be discussed offline

closed 2018-02-09 by DG9BFC