df8oe/UHSDR #1430: Ver. 2.8.0.

S52CQ, 2018-02-09:
Hi,

I am testing this SW ver. on mcHF. I've noticed that when I turn Off Waterfall (long press SPLIT, short press MEM) then also CW tuning help does not work. Also S meter does not refresh signal - it stays as it was when waterfall was on. Is it supposed to be like that?

73 de Jure, S52CQ

sp9bsl, 2018-02-10:
Hi,
For S-meter: this is because dBms are calculated in spectrum/waterfall state machine, when there is no waterfall or spectrum the state machine is switched off (same when you enter the MENU).  Looks like tuning helper needs it too. Will look at it in free time.

73 Slawek

db4ple, 2018-02-11:
Hi,
yes, if you disable the waterfall / scope the tune helper doesn't get the required data (as it relies on the data). While I don't know why one would switch off both waterfall and scope, it is quite simple to keep the processing going if the tune helper is turned on AND visible.

sp9bsl, 2018-02-11:
Hi Jure, 
thank you for this issue because I had few weeks ago problem with some setting while in menu and the s-meter didn't work when needed. So this was a motivation ;-)
Please test (#1442).

73
Slawek

S52CQ, 2018-02-11:
Hi Slawek,

no need to thank me - I was just trying to help with testing :) The community has to thank you all coders for doing great job!!

Will test new SW and will let you know.

73 de S52CQ, Jure

S52CQ, 2018-02-16:
Hi Slawek,

I am now testing 2.9.6. The mentioned bug is gone, but you will notice that S-meter is much more responsive when waterfall / scope are off. CW tune helper does work now.

I have noticed however that when waterfall / scope is off the Menu button below the LCD does not work.

73s de Jure, S52CQ

sp9bsl, 2018-02-16:
Hi,
Yes the readings are now much faster because we do not have massive core load due to lack of screen redraw in this mode.
I set spectrum and wfall 1/Speed to OFF (both DISABLED on the screen) and I can use the Menu button like before, no problem at all.
My lcd is 480x320 and rig is mcHF with 192kB/1MB.

73
Slawek

sp9bsl, 2018-03-01:
Hi,
can anyone confirm this?

73
Slawek

df8oe, 2018-03-18:
can be losed now?

S52CQ, 2018-03-18:
Sure, close it.

73s de S52CQ

closed 2018-03-18 by df8oe