df8oe/UHSDR #1437: 4.0.0 bootloader issue.

WM0I, 2018-02-11:
New bootloader now has a white screen when booting mchf to run the machine. After a second or two and then it show the normal boot screen. I retired to reload the 4.0.0 to see if that fixed the white screen problem but it didn't.

Regards Richard WM0I

df8oe, 2018-02-11:
I cannot confirm. Works as expected on OVI40 and mcHF.

df8oe, 2018-02-11:
Update: In the meantime I got many confirmations all is working fine...

OK1FWG, 2018-02-11:
I just upgraded my bootloader to 4.0.0 on a mcHF v.7 and if it is meant, that when shortly pushing the Power button, a complete white screen is lit, followed by a normal booting process. If it so, I am having the same behavior.

UR7FM, 2018-02-11:
Just upgraded to 4.0.0 from 3.4.2. On my RS918 same behaivor was with old bootloader... white screen for less as 1 second , after normal booting. I think its not problem.

df8oe, 2018-02-11:
white screen for < 0.25s is absolutely normal - no bug.

closed 2018-02-11 by df8oe