df8oe/UHSDR #1438: In AM mode AGC  "BUG"

ly2bok, 2018-02-11:
The AGC in AM modulation does not work. If the level of the signal is applied to the receiver input from the generator (-80 dB), and then the signal is turned off then the receiver does not return to normal reception. It remains locked at that level and there is no Rx  sensitivity. Even when changing to another modulation, the AGC "does not let go." Only "Power" helps.
Justas LY2BOK

DD4WH, 2018-02-11:
Hi Justas,
I listen a lot in AM and SAM and never had that problem. I just checked and cannot confirm.
Please let us know some more details:
* hardware type & display
* filter bandwidth
* NB setting ?
* NR setting ?
* autonotch ?
* AGC settings: for AM/SAM, I highly recommend the standard setting: vLO, which means a very slow AGC!
* what is your AGC threshold?

DD4WH, 2018-02-11:
1 | 2 | 3 | 4
-- | -- | -- | --
  |   |   |  
  |   |   |  
  |   |   |  
  |   |   |

df8oe, 2018-02-11:
Cannot confirm here. All is working as expected....

ly2bok, 2018-02-11:
https://youtu.be/U9EVvJgp-s0
Carefully watch the video. Settings do not change. The level of the signal is SSB, then I switch AM. I return back to the SSB where the signal is much weaker. I turn off the TRx. Again I turn it on. The signal of the SSB is again in a high level ... Agree, it should not be so. All the standard setting. On the other two TRx, that's the most. Ver. 0.4. STM  with more memory 1 MB.
Justas 73!

DD4WH, 2018-02-11:
. . . and concluding from your video it seems absolutely normal behaviour because you have switched on the Hang function in AGC.
Please have a look here:
https://github.com/df8oe/UHSDR/wiki/Automatic-Gain-Control-(AGC)
there you will find an explanation of the hang function and how to set the parameters.
All the best 73s
Frank

ly2bok, 2018-02-12:
I translated the letter from LY3BD Aleksandra. He broadly understands this. I do not know how accurately Google translates, but I think it will be clear. I also enclose the original text in Russian.
In the latest versions of the software there is one very unpleasant bug. If, in the AM mode, a sufficiently powerful signal comes to the transceiver input, for example with a level of 9 points or more, or some kind of impulse even a short-term noise ("click"), the receiver becomes "deaf". The AGC system processes the incoming signal, removes the gain, but after the loss of this signal, the AGC does not restore the gain of the receiver. AGC "sticks" and as a result weak signals are simply not audible. Now if you go to SSB, the receiver also remains deaf. You can exit this position by simply sending a powerful signal to the input of the receiver in SSB mode. Or turn off the transceiver and turn it on again, but it does not always help. Those who wish can make such an experiment. On any band, where there are powerful signals (9 points or more), disconnect the antenna from the transceiver, set the maximum gain on the HF by the AGC encoder. You will hear your own noises from the receiver. Turn on the AM mode and connect the antenna or send a signal with a level of 9 or more points. Now disconnect the antenna and you will not hear any of the receiver's own noise in either AM or SSB. The receiver became deaf! Now if the signals to the input of the receiver come with a low level, scores of 4-7, you simply will not hear them. You can exit the position if you send a strong signal to the input of the receiver in the SSB mode, with a level of at least 9 points. Then the AGC system will "wake up" and restore the sensitivity of the receiver. But as soon as you move to AM, the picture will repeat itself. None of the manipulations with the settings of the AGC modes in the menu do not give anything. The same picture is observed in the transceiver at LY2BOK.
In AM and FM modes, the AGC system must monitor the carrier level and not the level of the low-frequency modulating signal. Tracking the level of the modulating low-frequency signal in amplitude modulation and FM for AGC operation is not correct. The LF signal level is used for AGC operation only in CW, SSB and digital modes modes. This is a classic.
LY3BD Aleksandr
************************************************************************************************
Перевел письмо от LY3BD Aleksandra.  Он шире обьсняет это  . Не знаю как точно переведет  «Google» но думаю будет понятно.  Прилагаю также оригинал текста и  на русском.
В последних версиях ПО есть один очень не приятный глюк. Если в режиме АМ на вход трансивера приходит достаточно мощный сигнал, на пример с уровнем 9 баллов или больше или какая нибудь импульсная даже кратковременная помеха ("щелчок"), то приёмник становится "глухим". Система АРУ отрабатывает пришедший сигнал, убирает усиление, но после пропадания этого сигнала, АРУ не восстанавливает усиление приёмника. АРУ "залипает" и как следствие слабые сигналы уже просто не слышны. Теперь если перейти в SSB, приёмник также остаётся глухим. Выйти из этого положения можно только подав на вход приёмника в режиме SSB достаточно мощный сигнал. Или выключить трансивер и включить снова, но это помогает не всегда. Желающие могут сделать такой эксперимент. На каком либо диапазоне, где есть мощные сигналы ( 9 баллов и более) отключите антенну от трансивера, энкодером AGC установите максимальное усиление по ВЧ. Вы услышите собственные шумы приёмника. Включите режим АМ и подключите антенну или подайте какой либо сигнал с уровнем 9 или более баллов. Теперь отключите антенну и вы не услышите ни каких собственных шумов приёмника ни в АМ ни в SSB. Приёмник стал глухим! Теперь если на вход приёмника будут приходить сигналы с не большим уровнем, баллов 4-7, вы их просто не услышите. Выйти из положения можно если в режиме SSB на вход приёмника подать мощный сигнал, с уровнем не менее 9 баллов. Тогда система АРУ "проснётся" и восстановит чувствительность приёмника. Но как только вы перейдёте в АМ, картина повториться. Ни какие манипуляции с установками режимов АРУ в меню ни чего не дают. Та же картина наблюдается и в трансивере у LY2BOK.
В режимах АМ и FM система AGC должна отслеживать уровень несущей а не уровень низкочастотного модулирующего сигнала.   Отслеживать уровень модулирующего НЧ сигнала в амплитудной модуляции и FM для работы AGC, это не правильно.  Уровень НЧ сигнала используется дла работы AGC только в режимах CW, SSB   и в цифровых модах. Это классика.
LY3BD  Aleksandr

DD4WH, 2018-02-12:
Hi Justas, hi Aleksandr,
thanks for the detailed report! I will make this experiment later this evening.
Two points:
1.) as I already told you, please provide your AGC settings! This is vitally important and if you do not do this, I will not investigate your issue any further
2.) the AGC is programmed in a very "classic" way: taking into account the carrier level, because DC components are taken care of AFTER the AGC.
All the best 73s,
Frank

ly2bok, 2018-02-12:
•	hardware type & display --  v.2.9.2    LCD parallel
•	filter bandwidth   ---    in AM or SAM   5.0 ,3.2 Khz
•	NB setting ?        …..  OFF
•	NR setting ?        …….OFF
•	autonotch ?          …. OFF
•	AGC settings: for AM/SAM,  vLO, ----  And the rest is not affected
•	what is your AGC threshold?   ----0   to 120
•	STM    mem---- 1MB
•	Ver .M0NKA  0.4

LY3BD Alex

DD4WH, 2018-02-12:
Hi Alex, hi Justas,
thanks for your details!
I have tried to reproduce, but I cannot confirm your findings here:
1.) the video:
* after rebooting: in LSB on 80m the AGC threshold setting is much too low! There is no AGC action taking place as you can see from the AGC box, which does not have the sign "AGC" --> increase AGC threshold, have a look into the Wiki how to properly set the AGC!
* on the 41m BC band, you can see that there is the hang AGC enabled which makes the AGC stick for a few seconds, that is exactly the intended behaviour of the AGC --> switch of hang enable if you do not want that
2.) description by Alex:
I did the following:
* strong station in AM (Greece 9420kHz)
* disconnect antenna --> high noise level, internal noise hearable
* change frequ to a free frequency (9430kHz) --> no station present
* connect antenna --> high noise level
* change to 9420kHz --> same audio level again
Am I missing something here? Please try to describe what you do step by step and do not mix it with any interpretation of the radios internals behaviour, maybe I can then understand what your problem is. 
73 Frank

ly2bok, 2018-02-13:
https://youtu.be/hX6duU5ZTHQ
Once again I show. After the signal is sent to the antenna on the AM, the AGC is activated, and weak signals in other Rx modes no longer pass. This is not affected by any AGC prompts. This is very well audible in the video.
This occurs only in AM and SAM modes.
On another we can not explain......

ly2bok, 2018-02-13:
Good day, Frank!
By common forces, it was possible to determine that the hang of AGC in AM occurs only when the passband of the filter is more than 4.8khz. If the filter bandwidth is 4.8khz and less, then there is no problem with AGC. Please pay attention to this and, if possible, eliminate it. Thanks for your work.
LY3BD Alex, LY2BOK Justas.

DD4WH, 2018-02-13:
Hi Alex, hi Justas,
thanks for your investigations!

I have tested for an hour now: Sorry to say: but I really cannot see any difference in behaviour in my machine (OVI40 UI & mchF RF v0.4) using filter bandwidths below or above 4k8 . . .

However, I have spotted a small bug in the AGC, connected with the calculation of AGC slope. But since you have set AGC slope to zero, this bug will not affect the behaviour of your receiver. I will correct that bug in the next pull request.

Also, I still cannot understand the point what you think is wrong . . . we have a communication problem.
In your last video I do not see any problem with the AGC:

1.) you receive in LSB with the AGC threshold setting much too high!!! Please set the AGC threshold to a value so that the AGC box does NOT indicate AGC action when only noise is present in the filter bandwidth. Otherwise the AGC will do all sorts of things. Why would you want the AGC to boost up your noise signal? That is what you provoke if you set the threshold too high.

Please refer to the Wiki to set the AGC to the right values before your experiments. 

* search for a free frequency --> band noise
* use encoder 2 to adjust the AGC threshold ("AGC"), so that the white text "AGC" in the blue AGC indicator box starts to disappear
* With button M2 toggle focus of encoder 2 to next box
* Adjust AGC mode according to your personal preference and your band situation: OFF = AGC off
FAS = fast
MED = medium
SLO = slow
LON = long
vLO = very long

For broadcast stations use LON or vLO. Set slope to something different from zero (maybe 5dB), setting it to zero could potentially cause problems too. Set AGC Hang enable to OFF.

If the AGC still does things that you think are wrong, please report them again here.

All the best, have fun with UHSDR,

Frank DD4WH

ly2bok, 2018-02-18:
We wait. What, you still found?
Justas

DD4WH, 2018-02-19:
Hi Justas, hi Alex,
please be patient.
I am still working on this and trying to accurately identify the problem.
Stay tuned!
73 Frank

ly2bok, 2018-02-19:
We understood Thank you.
73!   Justas  , Alex.

WB8HSL, 2018-02-27:
I notice this gain reduction also.  V6 FW 2.9.9.

I have observed it in both sideband and AM modes.

Cycling power resets receiver gain to normal.

Scott

DD4WH, 2018-03-01:
Thank you all for your observations!
I could now reliably reproduce the issue. It is not an operating error, but seems to me likely a firmware bug.
When we switch from 4k8 filter to 5k0 filter, the internal software sample rate is changed from 12ksps to 24ksps. Some of the internally used AGC parameters are dependent on the sample rate. So somewhere the switching does not reliably affect the internal sample rate switching and new calculation of the AGC parameters. If you reboot, the AGC is started and initialized with the right sample rate based calculations.

However, I have not yet found the place where this goes wrong. 

So please be patient a little more, the bug is identified -I think- but I will have to find the place where something does not switch, but it should switch.

All the best 73s and keep on searching for bugs in the firmware! ;-)

Frank DD4WH

DD4WH, 2018-03-01:
Scott, thanks for your observation that this bug is present in SSB too! That corresponds well with my hypothesis, what the cause of the bug could be.

73 Frank

WB8HSL, 2018-03-01:
Frank,
You are very welcome.  If you need the full list of settings of my mcHF please advise.  Otherwise I won't clutter the conversation with that data.

I very much appreciate the work you and others are doing to make this little radio so enjoyable and so awesome.
73,
Scott  NM8R

WB8HSL, 2018-03-06:
DD4WH,

Is there any update on this issue?   Please advise if you need my operating parameters.

Thank you,

Scott  NM8R

db4ple, 2018-03-06:
@DD4WH: Does _changing_ the AGC setting via encoder E2 after the problem occurs, brings back the audio?

WB8HSL, 2018-03-10:
DP4PLE,

After observing for several days, it appears changing the AGC via the E2 encoder does not restore proper AGC action. Scott. NM8R

df8oe, 2018-03-12:
I never have had any issue since this report is active. Everything works as expected...

WB8HSL, 2018-03-13:
I am experiencing the same issue reported by LY2BOK regarding AGC hanging up.  Strong signals will cause the AGC to hang up and it does not release, desensing my OEM V6 radio.  

The only solution I have found is to cycle power to the radio.

ly2bok, 2018-03-13:
To: WB8HSL
( DF8OE : I never have had any issue since this report is active. Everything works as expected...)
No , not working AGC in AM . Working only very narrow filter. I have already lost hope that this problem will be eliminated. It's a pity.
Justas LY2BOK

DD4WH, 2018-03-14:
"I have already lost hope" 
--> well, that is your problem, not mine. ;-)

"very narrow filter"
--> 4k8 is not that narrow, or is it? ;-). The problem ONLY occurs with filters > 4k8 or do you have other experiences?

This is a low priority bug, so this will eventually be worked on, if time allows.

All the best 73s,
Frank

WB8HSL, 2018-03-14:
I will observe what filters it de-senses in, and report back.

While perhaps low priority, it is a debilitating bug.  I think no user actually expects the RX gain to drop and only cycling power restores it.
 I will gladly paypal for a quantity of favored beer to the programming guy who can fix it.
Probably LY2BOK would pitch in, too.
ScottNM8R

DD4WH, 2018-03-14:
"a quantity of favored beer to the programming guy who can fix it."
--> oh, now that could make a difference ;-). Greatly acknowledged :-).
No, just kidding, I will try my very best with this bug, but the AGC is a wild beast . . .

WB8HSL, 2018-03-15:
DD4WH,

As a fellow German, I make that offer with sincerity!

Best 73,

Scott

df8oe, 2018-03-15:
This "bug" never meets me. For me mcHF is an amateur radio - and no amateur works in AM. Only SSB, CW and digital modes. Additionally I never use filter bandwidth > 2.9K. So I never will meet this and for me it is no bug.It is dependent on your user profile... All programmers who want and will spend their time are welcome! But as long as there is more work than time to do it existent contributors set their own priorities regarding the most-used user profile. And I do not think BCL listening is a most-used user profile.

DD4WH, 2018-03-15:
maybe BCL listening is not the most-used user profile, but I am convinced that we have (and are in need of!) a rich diversity of different ham profiles, some of them using AM and SAM modes extensively for whatever purpose (BCL etc.). And thats a good thing ;-).
 
An amateur should/must have the freedom to use any mode or filter bandwidth he/she wants to use (of course respecting legal restrictions, thats a prerequisite). 

I had a look at the AGC problem again and I found an issue and tried to fix that, however my hardware is not working at the moment, so I submitted a pull request without testing it in the real world, leaving the job of testing to you :-).

df8oe, 2018-03-16:
That is absolute ok! But I "am out" - because I cannot reproduce this bug in short tests (regardless of signal strengths and / or filter settings) and it is not my working condition to listen hours or days to radio stations. So if I cannot reproduce in short tests others must do the job who are listening for longer times to radio stations...

DD4WH, 2018-03-16:
the newest software version of this morning contains a potential bugfix for the problem  #1475 .
I could not test it, because my hardware is in repair at the moment, so please test with your machines.
73 Frank DD4WH

ly2bok, 2018-03-16:
Unfortunately, there are no improvements in AGC's work in AM, SAM, everything as it was before. With a bandwidth of 5Khz and more, the AGC "hangs" and the receiver becomes blunt. But in the latest version of the program, you can restore AGC normal operation by simply switching the receiver's bandwidth to 4.8Khz. Previously, to restore the AGC, you had to switch the band to 4.8Khz, turn it off and then turn on the transceiver.
Aleksandr  LY3BD

DD4WH, 2018-03-16:
Hi Aleksandr,
thanks a lot for your test report, thats very interesting! Unfortunately I have to repair my mcHF before I am able to carry out further fixes . . .
Just one further question to you:
* you mean, with the last version you have set bandwidth to 5k, then the Rx becomes deaf and when you switch to 4k8 and then again to 5k, the Rx sensitivity is restored?
73 Frank DD4WH

DD4WH, 2018-03-16:
The newest version #1476 will contain my next trial to fix the problem. Could not test it, sorry!
Please test with your machines!
All the best 73s
Frank

ly2bok, 2018-03-17:
"Bug" corrected !
Instead of a lot of letters and conversations, everything is done. DD4WH Frenk, found an error, and the program is fixed .. So ... sincerely thank DD4WH Frenk. Now AGC AM works in the same way as the filters of different widths.
Frenk, thank you for the excellent work done.
  Yours faithfully, LY3BD Alexander and LY2BOK Justas

DD4WH, 2018-03-17:
Hi Aleksandr, Justas, Scott and everyone who participated here!
Thanks a lot for your efforts to test and report and your patience! Thats real team work!
Without you we would not have been able to identify this bug and finally fix it.
Keep on the testing and reporting!
All the best and have fun with UHSDR!
Frank DD4WH

df8oe, 2018-03-18:
can be closed now?

ly2bok, 2018-03-18:
Danke an alle .Die frage ist geschlossen.
Justas  LY2BOK  and Alex LY3BD

closed 2018-03-18 by ly2bok

WB8HSL, 2018-03-18:
I have been testing 2.9.16 over the last 24 hours in both AM and side band modes. 

I think the bug is fixed.

 Thank you very much, to all.  

(If I knew a paypal address I would wire some $ for a good German ale, as promised.)

Scott