df8oe/UHSDR #1439: ISSUE TEMPLATE FOR NEW ISSUES

DD4WH, 2018-02-11:
Hi folks!
issue reports are very helpful for us in order to improve the software considerably!
However, some issue reports lack basic information.

So I would like to introduce a structure for future issue reports which you can easily copy and deliver in every issue report for the UHSDR software:

HARDWARE | UI-board | Display size | Parallel/SPI | Tochscreen ?
-- | -- | -- | -- | --
  | OVI 40 | 480 x 320 | Parallel | yes

db4ple, 2018-02-11:
Hi,
a) yes, good idea.
b) there is a function in Github to provide issue templates. We should look into using this one. I don't think a voluntary template would solve the problem well enough:

https://github.com/blog/2111-issue-and-pull-request-templates

db4ple, 2018-02-11:
For the content: 
Mandatory version number firmware. 
Also for hardware (UI/RF).
Problem with requested data is that not all of your data applies to all issues and users get annoyed if they have to fill in useless data. We should have optional sections in the issue request (You have trouble with audio, please fill in the following data...).
Don't start with too much, we can add frequently required information as we go.

DD4WH, 2018-02-11:
Hi Danilo,
fully agreed!
One point: there is no "useless data" ;-).
The point is that the user can easily fill the header table (hardware) ONCE and then fill in the other fields for each issue. Come on, thats a matter of seconds and not annoying at all . . . :-). And users (in most cases ;-)) get an improved software version in return, so that really pays back to be accurate in filling the issue template :-).

Andreas, can you add that template?

73 Frank

DD4WH, 2018-02-11:
HARDWARE | UI-board | RF-board | Display size | Parallel/SPI | Tochscreen ?
-- | -- | -- | -- | -- | --
  | OVI 40 v0.6 | mcHF 0.4 | 480 x 320 | Parallel | yes

df8oe, 2018-02-11:
Hi,

I absolutely agree to get issue reports in a more precise manner. I will take a look in the possibility to force that. We will be flooded with senseless issue reports if we do not act here....

closed 2021-02-19 by db4ple