df8oe/UHSDR #1450: Treble control stops audio (BUG!!)

DG9BFC, 2018-02-14:
Your firmware version: 2.9.4
Your bootloader version: 2.0.0 ... but does not matter here

### Hardware 
 * UI Board: mcHF 0.4
 * RF Board: mcHF 0.4

### Describe the issue:
the treble control stops audio when turning down below zero (exactly at -2)

You have audio related problems:
 * DEMODULATION MODE: LSB
 * Filter:  issue is in all filter settings
 * Noise blanker setting: 0 (regardless if NB is on or off)
 * Noise reduction: ON/OFF (regardless if NR is on or off)

You have display (also fill if audio problem) related problems:
 * Display resolution: 320x240 (old 2.8 display)
 * Display connected via: Parallel 
 * Touchscreen: Yes
 * Magnify: 1x 
 * Spectrum: Scope

the treble control does not work like in older version

2.9.3 is ok ... 2.9.4 is not

sp9bsl, 2018-02-14:
I can confirm this on mcHF 1MB/192k wit 2.9.4. With BASS=TREBLE=0 all is ok,

db4ple, 2018-02-14:
Hi,
thanks for this report, I think this will allow me to fix the issue quickly. Good job!

db4ple, 2018-02-14:
Hi,
if you change bass or set treble to a positive value -> does this work?
if you save -2 and restart, there is no audio, with -1 there is? Correct?

DG9BFC, 2018-02-14:
when saved with -2 after power on a very short noise burst ... then audio stops
when saved with -1 audio works
that is treble
bass setting is exactly reversed
means minus settings in bass do work
plus setting breaks audio (+1 works +2 breaks)

db4ple, 2018-02-15:
Hi,
I could not reproduce the issue on 2.9.4 on the OVI40 so please (once Andreas merged my pull request) check out if this fixes the issue on the mcHF. Too be honest I have no idea why it is -2 or +2 which causes the problem. It is difficult to debug if you don't have machine exhibiting the problem... but my fix fixes a problem we had in the code and it may be the root cause for this issue (or not).

db4ple, 2018-02-15:
Ok according to @df8oe issue is fixed by PR #1451. We may have close a longstanding source of interesting problems (without knowing what problems we heard about might have been caused by this issue).
Waiting for additional confirmation from original reporter (Sigi), then please close this issue.
Thanks again to Sigi for finding a way to reproduce the issue and putting me on the right track.

DG9BFC, 2018-02-15:
confirmed ... treble and bass now works and audio does not stop
issue closed

closed 2018-02-15 by DG9BFC