df8oe/UHSDR #1452: new error msg

WD8BXS, 2018-02-15:
I am getting an  interesting and new for me message 

Your firmware version: D2.7.72
Your bootloader version: 3.4.6

### Hardware 
Ver 7 mchf

On boot up I am seeing SWR resistor mod not completed.

I have never seen this error before. Never heard of this Mod before.
What is it indicating?

TNX  Chuck WD8BXS

db4ple, 2018-02-15:
Hi,
the software since very long checks the ADC output and decides based on that result if you have this mod or not:

https://github.com/df8oe/UHSDR/wiki/Recommended-Modifications#rf-04-n-012-rf-04-n-022-correct-unsymmetrical-adc-in-the-swr-measurement

And if it think you have that problem, it will tell you during startup. I get this warning for instance also if there is no RF board is connected...

WD8BXS, 2018-02-15:
Found it, mcu pin

closed 2018-02-15 by WD8BXS