df8oe/UHSDR #1456: Please save scope magnification with mode when changing bands

RightRudder, 2018-02-16:
This is a feature request.  When I have different modes on VFO A/B or on different bands for example I want to watch a large section of the 80m band for activity at 1x mag, but switch back and forth with 40m band where I am decoding RTTY at 16x mag it is a drag to have to keep changing magnification.  Can we please store the magnification when mode info is saved so when I change bands or modes the last magnification I used with that mode or band is recalled.

df8oe, 2018-02-16:
First: This is not a trivial task. Band memory places are hard-coded: any change is killing complete configuration storage. But you are right: ther MUST be done something like this in the future and so we all will loose our configuration when stepping to such a new feature. But because of we do not want to bother users more than one time with complete new configuration we firstly must collect other settings which possibly are useful to store with band - and extend the band storage places with memory places (I think 100 is enough) so that you can store many favourites :)

Second:
recalling when switching mode OR band is not possible at the same time. And I do not like that feature - because my working condition is quite different from yours. So we must discuss a user friendly configuring possibility.

RightRudder, 2018-02-17:
These are good points.  What about some on/off type menu settings to allow user choice of what gets saved according to the operator's style?

df8oe, 2018-06-26:
this is within #1058

closed 2018-06-26 by df8oe