df8oe/UHSDR #1457: Please add one or two more filter selections for ssb mode

DG9BFC, 2018-02-16:
i did not fill in the form cause it is a general asking for all firmware and hardware versions
i have two of the four filter settings already used for digi modes (300/500 for psk and 300/950 for rtty)
so there are only two filters left for ssb mode (yes i know with long press i can get to all filters and then select another one ...but that was not the question)
if not to problematic (eeprom storage, arrangment of the default values in menue etc.) then i would either like two additional settings only for digi modes (filters above switched in when selecting that digimode ... maybe another set of four only for digi??) ... that would free up the two now used ssb filters ...
or two additional filter setting for ssb (then i can still use the lower two for digi and have four for phone in ssb
if you need some storage room ... i am guessing that we can use two cw filter settings cause for cw two filters (a narrow and a wider one) should be enough for normal work
but using two of the ssb filters for digi and then have only two left for phone ... that could be a bit improved (personal view ... a hard core cw user maybe thinks different)

df8oe, 2018-02-16:
My suggestion:

Hard-coded filters to every possible digimode - no user-selectable filters. Because of demod of digi signals filters, too there is no need for working-conditions depending additional filters...

DG9BFC, 2018-02-16:
yes that would be an option (as i already noted) that would then free up the now two used ssb filters for rtty and psk ... and then no additional ssb filter would be needed
so ... for psk a narrow (300)  filter with centre 500 ... for rtty same but centered around 1000 (i now use the 950 filter so maybe no need to add a new one)
the the better the filter does match the needs (as narrow as possible but as wide as needed) the better will be the performance
a narrrow filter does not let in so much noise and unwanted signals as a wide one (seems logical ... right?!?) ... even if the decoding stream already have a narrow own filter ... for the user to tune in and hear a tiny signal a narrow filter is a must
what we then later could try is if adding the peak filer or using the noise reduction or noise balnker does disturb ... or help the digital mode used ....
some signals may be improved with lowering the noise .... and the noise reduction now works superb
so some more testing to be done :-)