df8oe/UHSDR #1460: Suggestion : realized a tool to compare menu setup

iw2fiv, 2018-02-20:
Every time we update the fw the routine create a backup of menu's configuration.
I think that an external tool to read and compare this back up file  with another may be helpfull to compare for example a config with a friend.

df8oe, 2018-02-20:
Not the configuration is backed up - it is the firmware itself (without configuration). So a comparison is without any use.

frazierjason, 2018-02-22:
I gave some proposals in #1344 that might fool this request. I haven't started working on them yet.

db4ple, 2018-04-20:
You can now save the UHSDR configuration via USB to a PC using the uhsdr_tool.py (or on Windows the uhsdr_tool.exe). See here: https://github.com/df8oe/UHSDR/wiki/Backup-and-Restore

It is not exactly what you have in mind, but as close as it gets right now. 

@df8oe : Please mark this this as duplicate of #1344 (or vice versa).

df8oe, 2018-04-29:
If you are satisfied with the informations: please close this issue. Thank you!

iw2fiv, 2018-04-30:
No it's not what I have in my mind.
I try to dig in the source where the menu items are generated and coupled with the relative variable actual value.
I don't know if it's possible or it's sciencefinction but that is the possibility. It's true that each time fw is upgraded the "parser " need to be updated.