df8oe/UHSDR #1465: Version S2.9.10 hangs/locks up. Happens when mode is changed to SAM and if DSP is activated. Teted on my clone, 512MB.

lb5va, 2018-02-27:
# UHSDR Firmare Defect Issue (Bug) Template 1.0

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Your Issue Data goes here:

Your firmware version:
Your bootloader version: 

### Hardware 
 * UI Board: mcHF 0.x / OVI40 1.x
 * RF Board: mcHF 0.x / OVI40 1.x

### Describe the issue:

Place tell us what does not work and how others can reproduce the issue (list related settings, steps to do, etc.):

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

You have audio related problems:
 * DEMODULATION MODE: LSB/...
 * Filter:  2K7 LPF
 * Noise blanker setting: 0
 * Noise reduction: ON/OFF

You have display (also fill if audio problem) related problems:
 * Display resolution: 320x240 / 480x320
 * Display connected via: Parallel / SPI
 * Touchscreen: Yes/No
 * Magnify: 1x / 32x
 * Spectrum: Waterfall/Scope/Dual
 
Other settings you consider as relevant:

 * ...

db4ple, 2018-02-27:
That is quite mysterious, we need more reports to confirm the issue happens on multiple machines. In any case, the change between 2.9.9 and 2.9.10 only affects STM32F4 based machines It is a very, very minimal changing in timing so I am curious how that could have such an effect. 
@lb5va: Is there anything connected via USB (CAT, Audio), when the problem happens?

jelafond, 2018-02-27:
My original kit from Chris locks exhibiting exactly the same behvior as described by Ib5va. Running parallel LCD,   no IO connected,  USB and LSB,  Filter in default,  Blanker 0,  nr is off,  Touch active,  Mag x1, All three Spectrum displays tested.  No Mods on vers 6 kit board with STM changed out to the '429. Did a full reset of Rom after Flashing to bootloader v4.0.0   -- Back to UI v3.9.9 and all OK.  john k0vfr

df8oe, 2018-02-27:
I can confirm radio locks if I switch on DSP or mode changes from SAM to FM (regardless of other settings - 100% reproduceable)

STM32F427

vy 73
Andreas

db4ple, 2018-02-27:
Hi,
this is quite mysterious to me, How can such a change cause this kind of problems. Unfortunately I don't have access to a mcHF for debugging at the moment.

WB8HSL, 2018-02-27:
My OEM V6 became very very slow to respond to button pushes after installation of 2.9.10 last night.   It could potentially be interpreted as locking up as the delay became that long.  I rolled it back to 2.9.9 and all is well, again.  Scott

WB8HSL, 2018-02-27:
Clarification:  This was in AM mode, as well as SAM.

Scott

db4ple, 2018-02-27:
@df8oe:
I recommend to revert my last commit until I can perform serious debugging and identify what is the problem here.

quadrant2005, 2018-02-27:
I can confirm, 2.9.10 firmware is locking up radios v0.4 and v0.6 i just updated because i was looking into getting a keyboard and mouse working, on version 2.7.0 the keyboard worked but found on 2.9.9 that keyboard was not working i was not sure if this was my change by adding the break in my own firmware version so installed your latest build that does not have any of my changes and it says Firmware change detected  Please review settings!  ....Starting up normally... but gets no further that is on a v0.4 board with a 2mb nand flash processor, on v0.6 with 1mb nand flash radio it very sluggishly reaches the interface but hags with no updates.. Hope this info is helpful in some way.. regards Richard 2E0DVK

quadrant2005, 2018-02-27:
please how do i remove the last commit from my fork as ive commited your changes thank you regards Richard

db4ple, 2018-02-27:
DF8OE will do this on the official branch. No need to worry. If you temporarily want to go back to a previous commit you can use either git reset -- hard HEAD~1 (replace 1 with the number of commits you want to go back). This lets your local git "forget" all newer commits. Or you use git checkout <commit_hash> which will just send you back to a commit with the given commit hash, see for instance git log to find the hash value. This will not let git forget anything just brings your working directory to contain the selected commit.
Or you wait until DF8OE does the magic of reverting a commit with a new commit.

quadrant2005, 2018-02-27:
ive comited and REVERTED cheers :) thanks db4ple hope you guys get things sorted ok..

quadrant2005, 2018-02-27:
p.s if its helpful at all in the future i have a v0.4 512k a v0.4 2mb  and v0.6 1mb McHF...  the 2mb is my dev radio

db4ple, 2018-02-27:
I now know what the problem is (not yet how to solve it): Long time ago I noticed that we get into trouble if the clock configuration of the bootloader and the firmware differ. With my commit we recreated that situation.  That explains why a on its own correct clock configuration does not work as expected. I guess we need to change the way how we leave the bootloader so that a later clock configuration works as expected. Will look into this.

db4ple, 2018-02-27:
I just made a new pull request (#1467) which should by all means solve the not working clock reconfiguration issue by bringing the clock configuration to its "reset" state before applying the new configuration.

quadrant2005, 2018-02-27:
mmm sorry can not undo the revert on my fork to be able to try your change grrr... just tried

quadrant2005, 2018-02-27:
got it reverted the revert hehe will make your change now and try on my code base

quadrant2005, 2018-02-28:
TESTED on v0.6 McHF the above BugFix worked ;-) sorry it took me so long to come back, did me first build with the ovi40 main open and it did not like it :+1:

PR #1467 (2018-02-28): Fix for clock reconfiguration
This change should enable reconfiguration of clocks in firmware
so that we can use a different clock configuration in bootloader
and firmware.

This should fix #1465. Please test, I don't have access to hardware right now.
closed 2018-02-28 by df8oe