df8oe/UHSDR #1466: Version S2.9.10 hangs/locks up.

lb5va, 2018-02-27:
Happens when mode is changed to SAM and if DSP is activated. Tested on my clone, 512MB. 
Screen and all freezes and sound also gets funny.  Spectrum form in 2x looks a little strange. Lower levels on right side. 
Back to S2.9.9 all is ok.  Tried it all two times, included new download. 
Stig

db4ple, 2018-02-27:
Hi Sti,
please fill out the relevant details in form in #1465 and close this duplicate issue. Thank you.

jelafond, 2018-02-27:
I also experienced this as well on Version 6 Original Kit. The MCU has been changed to the 429.
Works well under S2.9.9 There was no IO at the time of lockup. Press DSP twice and locks every time.
Yes - its timing me thinks.  Some optimizing will solve. 
john  K0VFR

df8oe, 2018-02-27:
duplicate #1465 - closed.

closed 2018-02-27 by df8oe