df8oe/UHSDR #1469: Lost everything by any chance did anyone download or fork my code base

quadrant2005, 2018-03-01:
Totally a long shot but did anyone download my code base, lost everything the 2.9.10 pull got me in a mess.  I recovered my fork, but applied the bugfix that db4ple posted this fixed the code base, then when 2.9.11 of the main fork went up, i could no longer merge, i removed the changes in my file but still would not let me merge.  I moved my work into another folder, in the end deleted my fork and reforked, then sync to my computer then ment to copy my work into the UHSDR folder so git desktop would see my changes and intergrate them with my fork but i messed up i copied the folder the wrong way and over wrote my work, I had rar the folder before doing this but once again it was your code i backed up not mine. I hate being sick it really sucks :( lost everything.. I can not believe i did it,  Ive managed to pull changes upto the 14th of feb from this pull request, but lost 14 days work, i got two devices working on the HOST port lastnight it requires some changes in one of the files because is it cubemx or something doesnt handle the second host it put it all in one structure and gets over written, i had working keyboard and mouse lastnight so so totally gutted...  i made a massive code push yesterday its my fault but feel so upset....   I did keep backups of all my work but on a usb pen drive which got damaged the other day because i dropped my laptop and it was plugged in the side, Just having a bad run of things at the moment here...   totally my fault but feel so down and gutted right now.. I need to setup some kind of automated task on my machine so that it keeps autobacking up my work to dated folder or zip etc so that i do not end up in this situation in the future,

db4ple, 2018-03-01:
Hi Richard,
git does not forget easily. Please read the following web page: http://effectif.com/git/recovering-lost-git-commits. With `git reflog` you will be able to restore previous states of the repository. 
However, if you destroyed the repository itself, then git reflog may not be able to help you to recover.
But it is worth a try.

quadrant2005, 2018-03-01:
hi there db4ple, yes i was not thinking clearly and deleted the fork. I even tried using the git commandline to merge it, But i did not manage to fix it so i deleted and recreated the fork, my intention was to sync with the head fork, then swop out the files locally for my own, but i also was not thinking clearly on this because it would see all the files as changed, I was even stupid enough to backup and rar the pull from the head fork instead of my own code base, what was i thinking... Ive used the push request i made here that submitted my code changes, from there i was able to get my code upto 2.9.4, i then wrote all the changes down in a txt file. downloaded the latest 2.9.11 src from the head fork then intergrated my work, Ive lost quiet a few things though,  With being very unwell im making stupid mistakes so much lately. The thing that im most upset about is i managed to get the HOST port to enumerate two devices, the cubemx or what ever it is called creates an issue in a certain function so the HOST device gets over written, i found information and code on the net about this and got it to treat HOST devices one by one, i had my logitech k400 keyboard/mouse working and had got as far as printing the mouse location to the screen. Im not going to give up i will try harder, i now have a backup folder once more for each release version and also have created a BACKUP fork on my repository so that i can periodically sync it to my main fork.  Thank you for your help and your patience..  I want to write a function to backup the eeprom to the USB Drive and via versa, this could allow for settings to be copied across radios in theory. i need to remember to enable usb drive access as it is currently only enabled in bootloader i saw. Bye for now a little rest time now...

quadrant2005, 2018-03-01:
found the page about it here :) was still open in my browser pages ive 93 open hehe
https://community.st.com/thread/46444-stm32-hid-host-class-library-mouse-mode

db4ple, 2018-03-01:
BTW,
you should NOT use merge/pull to integrate changes happening df8oe's `active-devel`. It is in a way better to use "rebase" as documented on our contribution page. We had very little trouble to follow that approach for the last two years. It also has its issues but in general, once you understand how to operate it, it works quite well. The main benefit of the rebase approach is that it keeps all your "local" changes together on top of the last official release you rebased on.  Opposed to the merge approach you have been running, which does mingle your changes with the external changes in the time history.

df8oe, 2018-03-12:
We cannot serve any help here. So if noone has cloned your repo I think there will be no solution - issue can be closed?

closed 2018-04-11 by quadrant2005