df8oe/UHSDR #1470: NR-NOTCH problem. Slows everything down.

WM0I, 2018-03-03:
## Your Issue Data goes here:

Your firmware version:  2.9.12
Your bootloader version: 4.0.0

### Hardware 
 * UI Board: mcHF 0.x / OVI40 1.x    Clone
 * RF Board: mcHF 0.x / OVI40 1.x    Clone

### Describe the issue:

Place tell us what does not work and how others can reproduce the issue (list related settings, steps to do, etc.):    When pushing the DSP button to get to the PEAK (755 value) I have to go through the NR-NOTCH setting. This is where the problem comes in. Every thing slows down on the scope to a crawl. all the buttons and all the knobs are not responsive. I have to keep tapping on the DSP key to get it off the NR-NOTCH setting so that I can get back to normal scope and button response. 

I think what is happening is the CPU is not fast enough to handle all the buffers to do the new Noise Reduction and Notching.

Need away to get out of the NR-NOTCH function with having to tap the DSP button so many times to get to normal.

Best regards
WM0I

df8oe, 2018-03-04:
I can confirm this impacts only radios with SPI LCD. Activating DSP NR slows radio down a little bit, but activating AutoNotch, too makes it nearly unoperable. Working with manual notch does work.

Radios with parallel LCD are NOT AFFECTED - OVI40 not, too.

Question:
If not solveable via firmware because SPI LCD is the bottleneck: shall we disable AutoNotch together with DSP NR when running on a radio with SPI LCD?

vy 73
Andreas

db4ple, 2018-03-04:
Hi,
that is good and bad news together: It shows that the High Priority Task concept delivers uninterrupted audio even under very high loads (since there was no report about choppy audio) but of course, it cannot do wonders if the overall performance is not sufficient. And the result of this is sluggish response of the UI.

Disabling Auto Notch only for SPI / STM32F4 is technically possible but of course calls for trouble in terms of support requests ("Why can I don't get AutoNotch plus NR to work?") even if we mention that specific behavior in the documentation. If there is no other solution for now, we should go this way. 
Let us think for a brief moment of other ideas. If nothing turns up by the next days, lets do the disabling of the combination.

DD4WH, 2018-03-04:
there are two things we can/should do:

1.) alter the way the DSP box works. Let us think about a way to choose the different options without having to toggle through all the modes. Also some combinations of DSP goodies are not yet available at all (for example peak + NR, or manual notch + NR which could be very useful!)
--> so we need to do the following:
* enable autonomous switching of manual notch filter, peak filter and maybe all of the DSP options? By the way: "DSP" is a totally misleading name for that switch, because basically ALL of the things we do in the audio path is DSP, not only those few options ;-)
2.) we could try to lower the load for autonotch by changing the parameters for autonotch when SPI display is present. Eg. DSP_NOTCH_NUMTAPS = 32 and DSP_NOTCH_BUFLEN=64, maybe this already helps a bit . . .

73 Frank

df8oe, 2018-03-04:
Of course disabling this would be the last solution. But if no other possibility is found it *is* a solution.

Firmware is doing an superb job - but it cannot do wonders. If there is a bottleneck in hardware this cannot be removed by software. Running LCD in SPI mode _is a bottleneck_. It only can be removed by hardware changes.

I can confirm, too that audio is remaining in best quality at all so "task weighting" is running fine.

vy 73
Andreas

sp9bsl, 2018-03-04:
I would like to make the menu shown on the screen (for DSP box in this case) with the virtual keys to choose between options using the touchscreen and for those who do not have it enabled, to use encoder and a key. Idea is to display available options, and then accept one of it. So the most processing demanding options should be marked with warning (for example yellow triangle with exclamation) of slowdown possibility.
Off topic: this could open the discussion for new features with virtual keyboard like choosing band from list and direct frequency set or filter selection.

73 Slawek

DD4WH, 2018-03-04:
Slawek, that sounds really excellent and would mount the user interface into a new dimension!!! 
One small remark: I do not think we need the warning colors or something similar. People who have the small processor will try once and then they know what their options are :-).

db4ple, 2018-03-04:
Hi,
very much in favor of the virtual keys. If we use the "menu area" we don't have to worry about redrawing, we can simply draw over this area without too much issues. 
However, I am not thinking of a virtual keyboard (different thing, related but not same). I would be more in favor of a low number of large boxes which are a) easy to touch on and b) can also be selected using the encoders for those without touch screen (we could recycle the menu operating principle: M2 switches between buttons, M3 selects/deselects function). Selection or deselection should be indicated visually (change of color of selected item, a check mark before item, whatever).

sp9bsl, 2018-03-04:
Danilo, exactly: highlighted button (box) for selected option. Yes, the "virtual keyboard" is not the best description of this thing, but I thought of virtual keys :)
Im busy now but will start with this maybe next week or so...

df8oe, 2018-03-05:
My observations:
2.9.13 improves behaviour - but it is not satisfying.

My opinion:
We have reached edge of mcHF running SPI LCD. Small MCU in combination with waterfall and spectrum && DSP && autonotch is too much load. I think it is not solveable.

My suggestion:
Let us add everything which works and users must decide if they want to use these new functions (possibly in special combinations) with special mcHFs (small MCUs, SPI-LCDs and so on).

So we must open a new WIKI page called "known problems". I do not think we should name it "issues" - because it is not an issue, it is running out of hardware ressources. This page should list each problem which comes up on special device - so no issue report must be given to such a problem.

Additionally the idea of sp9bsl ("virtual keyboard") is a very good one. Adding warning signs to critical functions on special hardware is a good idea, too.

db4ple, 2018-03-05:
Hi,
okay. We'll leave that "fix" in for a while but consider removing it in future. But I will work on a different idea to improve SPI display performance in general based on the use of DMA interrupts. We'll see if that works out. If so, it should free considerable processor time for computation tasks as opposed to waiting for the SPI transfer to finish and by this the user interface should become more responsive on SPI machines. Eventually we could use this approach for parallel displays as well.  The idea is simple: For the redrawing of waerfall and scope, we break out the actual "main" drawing code so that drawing itself can be handled in an  DMA interrupt. As there is not much real computation, just drawing requests going on, in the interrupt which comes once the DMA buffer is empty, the DMA buffer is filled quickly again and then the slow SPI communication takes place while the processor handles other interrupts including the high priority tasks. Buffer is empty and no further drawing required, we go on as usual with the application code.

This is independent of the improved DSP function selection approach which should be done in any case.

DD4WH, 2018-03-05:
Hmm, thats a little strange, that the smaller filter does not help considerably:

* Danilo has changed the filter from 200 to 89 taps. So that means 111 taps saved in 48ksps
* the autonotch LMS filter has 64 taps and needs a little calculation time for updating the filter coefficients (but in decimated sample rate, which is in most cases 12ksps). So the equivalent of 64/4 filter taps = 16 taps (plus some LMS calculations) more (for autonotch) should be compared against 111 taps less . . 
--> So the smaller filter should definitely cure all the problems IMO, there should not be any sluggishness in encoders etc. any more
--> I would also argue (as Danilo said), it is not a pure problem of reaching the edge of the STM32F4, but a problem of the display drawing taking too much time for waiting. 
Danilo, if you can distribute the processing time a bit better among the different tasks, that would be excellent!

73 Frank

WM0I, 2018-03-06:
Seeing the writing on the wall I see I need to buy a UI completely populated board to be able to set the display to parallel configuration and get away from the SPI problems. Where do I go to buy one? 

Best regards
Richard WM0I

db4ple, 2018-03-06:
Richard, you should buy an OVI40 UI:
 a) mcHF in the V0.4 to V0.6 are not being sold anymore by Chris, M0NKA, you'll get them only second hand if at all.
b) the OVI40 UI boards is formfactor and electrically compatible to mcHF RF board.
c) it has a much more powerful processor and if you want a either larger and/or higher resolution display.

See here: https://www.electronicrepair.de/ovi40-order-page/

db4ple, 2018-03-06:
That being said, I still hope we can improve the performance of the UHSDR software SPI handling so much that we can run the SPI display with almost as little overhead as the parallel display. But I cannot promise anything here.

DG9BFC, 2018-03-14:
hmmm if i remember right the double display (spectrum and waterfall at once) needed a lot of cpu power and you deleted the old nr for adding the double view 
before i could use old and new nr combined and that was the best nr we ever had ... but thats another story and i can live with the new nr ...
ok back to topic ... can you delete just the double display for spi units?!? would that increase the speed somewhat for the spi units?!? (just an idea from a guy who has no idea what you clever guys are doing)

RichardPar, 2018-03-24:
Hi there,

Is the SPI Display really the cause? I am questioning as the unit I have shows 'HY28A/B Para' in the info and when I press the DSP button to NR+NOTCH it goes non-responsive. 

Richard 

Running D2.9.17 
[image]

df8oe, 2018-03-24:
That is related to the mode you are working on. it is in the nature of the physics, that modes with much higher system load cannot be used with NR&Notch activated.

This are the modes you can use NR&Notch activated using parallel LCD without impact:
- SSB
- CW

This are the modes you cannot use them at mcHF (but at OVI40):
- AM
- SAM

This are the modes where the use of NR&Notch is without any sense:
- digital modes
- FM

If you are running SPI-LCD, you cannot use this combination at all.

If you use OVI40 it is usable in all combinations (but of course in some it is without any sense).

73
Andreas

RichardPar, 2018-03-24:
Thanks Andreas,

I have a Parallel display and I cant use SSB with NB+NOTCH (A-NOTCH is fine) ... hence my question.  I think SPI/Parallel is a red-herring

Richard

df8oe, 2018-03-24:
We must clarify. Do you mean NB or NR? Please write as exactly as you can which combintaion deos not work. Maybe there are some more that are unusable at mcHF because of a lack of horse power.

Andreas

RichardPar, 2018-03-24:
I mean NR .. (Been playing with a Yaesu all morning and it has a NB)

If the bandwidth selected is above 2.5K, then the NR+NOTCH goes bad. (CPU overloaded... below that its still okay)

df8oe, 2018-03-24:
I cannot confirm. My mcHF with parallel LCD does work with 2.9LPF, NR+Notch perfectly (SPI does not).

But I agree: DSP stuff is pushing small MCU to its limits.

73
Andreas

RichardPar, 2018-03-25:
@df8oe and Others,

Is there a simple flowchart/diagram on the audio processing flows? I am trying to understand the flow in the audio processing. I am thinking about the realtime needs of a Notch filer.

Detect Tones <---
Apply Notch Filter
Audio output

The detect tones part does not need to be real-time, but can be run in 100mSec (example) intervals. The input data to detect frequencies for 'notching' can be scaled to 8 bit as it should still be sufficient to detect frequencies.

IT may already be like this :D - only been looking at code for a few hours (and getting a STLink debugger tomorrow)

Best Regards,
Richard

db4ple, 2018-03-25:
Hi,
I recommend to have a look at (https://github.com/df8oe/UHSDR/wiki/Topics:-UHSDR:-Theory-of-Operation) and especially on the NR page. And for NR, yes it is somewhat like this.
However, the current Auto Notch is still the "old" LMS filter approach which runs inside the audio interrupt and eats substantial horsepower there, a new approach is being worked on by @DD4WH / @dl2fw based on FFT etc. but not yet ready for primetime.

RichardPar, 2018-03-25:
@db4ple Awesome! a gold mine.  If @DD4WH and @dl2fw need a guinea-pig, then I am willing to be that tester :D

Richard

DD4WH, 2018-03-25:
Hi Richard,
Danilo already gave the link to the Wiki, there is a description of the audio path there. A diagram is not yet available, but will be prepared for a talk at the HAMRADIO SDR academy in June 2018.
There are different approaches to notch filters:
* manual notch: this is fully working in UHSDR with a very high Q IIR biquad filter, which needs very low processor power. So this is perfect to notch a constant carrier. This is already implemented.
* auto notch LMS (time domain): this is the autonotch already implemented and working very well, but quite hungry in consuming processor power
* auto notch frequency domain: not implemented yet. However, it will probably NOT save a lot of processor power, because it only works when the spectral noise reduction algorithm is active (FFT-inverse FFT audio chain). Additional problem is that you cannot simply notch carriers by deleting specific bins --> this introduces annoying audio artefacts (however we experiment with different windows which could help a bit). The idea at the moment is to detect carriers to be notched in the frequency domain and to notch them with dynamically adjusted IIR filtering in the time domain. 
I doubt that this will save many processor cycles . . .
The latter reason (and the fact, that the LMS autonotch is very good at killing annoying carriers) means that there is not a particularly high priority for us working on this. 
But if you are in for a trial to help here, go ahead! Its fun!
Best 73s, Frank DD4WH

df8oe, 2018-04-29:
If you are satisfied with the informations: please close this issue. Thank you!

closed 2018-05-27 by WM0I