df8oe/UHSDR #1473: NR problem

S52CQ, 2018-03-12:
Hi,

When testing last few builds I've noticed that NR behaves strange. Here is what I've discovered: I press DSP Button and NR shows a setting of 5 - and it would not let me set it to lower value than 5. Also BR distorts audio so it's not recognisable in times.

My settings:
HARDWARE
UI-board	RF-board: mcHF v0.6	mcHF 0.6
Display HY28A Parallel
Touchscreen: YES

DEMODULATION MODE: USB
Filter-BW	2.3 LP
Noise reduction ON
Autonotch OFF
DISPLAY	Magnify size 1x	waterfall
EEPROM: 24xx1026/CM01/128k
Flash size: 1024
RAM size: 192
Firmware: D2.9.14
Build: Mar. 11. 2018
Bootloader: Monka

Best regards
Jure, S52CQ

df8oe, 2018-03-13:
Hi Jure,

NR level is a long time at lowest setting of "5". I cannot remember when this started but it was some weeks ago. I do not have any distortions using mcHF and OVI40. I am working always with enabled NR since it is our new one - this would be an issue I see very fast....

I tested it using same settings as you - it works. So we need some other reports.

73
Andreas

WM0I, 2018-03-13:
Andreas, you had other reports from me when 2.9.14 came out but dismissed the report as operator error.  I need to know why when I loaded 2.9.14 I got all kinds of menu values changed. I didn't lose power when installing 2.9.14 as you suggested.  

73 WM0I

df8oe, 2018-03-14:
Hi Rich, there is no reason why you should loose any settings at 2.9.14. Noone of us can see any code changes that may cause this, many others are working this version without any lost of settings. So there must be a reason which is only present at your device. It can be poor hardware quality of clone - and an operational error. The operational error would not be at flashing new version but at the last power off cycle BEFORE you flashed new version.It is not possible to check it out but if it was so the issue would have been present with the installed firmware, too. But as I wrote: it is not possible to test this.

And again regarding NR: I cannot observe any distortions or different behaviour in NR (now I worked for one day).

S52CQ, 2018-03-14:
Hi Andreas,

yes, maybe it's HW issue... or other setting in the SW. I have RX IQ set 
to automatic correction. I did remove 4x LM386. That's all I can think 
of my head.
If I increase NR to No. 10 or higher then distortion is really bad, 
audio cannot be understood like person talking under water :)

Let's wait for other reports.

Best 73s de Jure

df8oe, 2018-03-14:
Hi Jure,

I do not have installed 4 LM386, too. The possibility of impacts from other settings is high.

The sound you are talking about is present and was present all the time. It is the only "negative impact" of new NR: the artifacts which sounds like waterfall :) They never were smaller than at actual firmware. Maybe you do not have recognized them before? Or you do not have tried to change settings of NR to smaller values as "5"? Please can you compare new firmware again with an older one where you think these phenomens are not present? It would be very helpful if we do have two versions which can be compared regarding this!

73
Andreas

S52CQ, 2018-03-14:
Hi Andreas,

I did recognize waterfall artefacts and I thought that this is still in 
development stage.

Sure I can compare older versions, I have them here and will report to 
you in a day or two. Then I will write a report of my SW settings.

I cannot set NR smaller than 5 - it just won't go lower. Is that a bug 
or a feature?

73s de Jure

df8oe, 2018-03-14:
It is a feature. NR can only be selected by +/- "5" steps.

df8oe, 2018-03-14:
And regarding "waterfall artefacts": That is a not-removeable impact of DSP stuff. But they are "melodical" and so do not harm my ears ;) We have compared NR of four projects (some expensive commercial products, standalone DSP like "Lingua") and found out that our NR is absolutely compareable to them. So this is not development stage - it is "final".

DG9BFC, 2018-03-14:
and it does a really well job :-)
you can dig out signals that are barely hearable
it does not work wonders ... but it is close

S52CQ, 2018-03-14:
Hi,
when comparing Ver. 2.0.0., build Mar 18 2017 and build D2.9.14 Built March 11 2018 I see in the V.2.0.0. possibility to set NR in steps of 1 and I can hear difference between 1 and 5, even if I set NR to zero, I can hear difference between no NR and NR=0. It's clear that there is less noise with NR however waterfall sound at NR=5 is at Ver 2.0.0. distinctive - nearly same as at the latest FW version.

But you're right, noise reduction is great. I do have terrible noise at 80m and NR does help a lot!

here are my settings:
AGC WDSP Mode: FAST
AGC WDSP Slope: 0 dB
AGC WDSP decay: 100 ms
AGC WDSP Treshold: 10 dB
AGC WDSP hang Enable: ON
AGC WDSP hang Time: 500 ms
AGC WDSP hang threshold: 45 dB
AGC WDSP Hang decay: 100 ms
RX Codec Gain: AUTO
RX/TX Xlate:RX-12 kHz
RX NB Setting: 0

73 de Jure, S52CQ

db4ple, 2018-03-14:
Hi,
the NR in 2.0.0 is completely different from the NR 2.914. Each of the different NR use  different parameter range. Don't compare them. Using a non-Zero value for minimal NR makes sense. It took me a while to figure out that NR = 0 and NR off are different things in the old firmware...

S52CQ, 2018-03-14:
Well tell me then which version to test/compare with and will do that.

73s de Jure, S52CQ

DG9BFC, 2018-03-14:
nothing to compare ... use newest firmware and have fun :-)

DD4WH, 2018-03-14:
since 11 Feb the NR has a new adjustment range of 5 to 199 (instead of 0 to 99, and a few months ago it was totally different :-)).
But that current range of 5 to 199 is not linear in its NR action, you will have noticed that (and to account for that, it is adjustable in steps of 1 in the high range and in steps of 5 on the low range).
A distortion with the lowest setting 5 has to be a hardware issue or some really weird setting., I think.
You also have the version with small RAM, so with that hardware you might not be able to use the new NR.
Did you try the small build?

S52CQ, 2018-03-14:
Small RAM? I thought that RAM No. that I've wrote was meant free RAM out of all RAM available... That was seen in the System information anyway.

Sure I can try small build and will report.

73s de S52CQ

df8oe, 2018-03-15:
192KB RAM *is* small RAM. In system menu you see the complete amount of RAM. On 192KB machines ~160KB are always in use and in the rest there resides an "unknown amount" of stack which is hardly to measure. So you do not have much RAM free for functional purposes. Firmware development is already at the edge of small MCU (RAM and flash, too). There is not much room for improvements and/or new features. But regarding NR I do not think this is RAM dependent. New NR is very powerful and I cannot see any issue - even at small MCUs.

df8oe, 2018-03-18:
can be closed now?

closed 2018-03-22 by S52CQ