df8oe/UHSDR #1479: Why is there an audio echo coming out the Line Out port?

WM0I, 2018-03-17:
Your firmware version: 2.9.xx - 2.9.16
Your bootloader version:  4.0.0

### Hardware 
 * UI Board: mcHF 0.x / OVI40 1.x clone
 * RF Board: mcHF 0.x / OVI40 1.x clone
### Describe the issue:

Place tell us what does not work and how others can reproduce the issue (list related settings, steps to do, etc.):

The audio is great when it comes out the internal speaker.
As soon as I connect the line out to my PC computer the audio becomes a strong echo. I use the line out for SSTV audio to the computer for decode. 

I have a Bluetooth USB dongle but the McHF clone has no provisions for bluetooth. 

There was no change to any of the settings at all.

### Your relevant settings

df8oe, 2018-03-18:
I don't think this is coming out the line out port. Maybe you do have activated your microphone input at your PC so that you mix up audio from line out to the one which you get via mcHF loudspeaker and PC microphone?

G3WKW, 2018-03-18:
I wondered why you would not be using the USB audio. Perhaps you have that connected as well?

Bob Thornton

WM0I, 2018-03-18:
I will have to ask as how to use the USB. Which port? What menu item do I use? Can someone point me to where I can learn about USB audio on the McHF?

Best regards
WM0I

G3WKW, 2018-03-18:
https://github.com/df8oe/UHSDR/wiki/USB-CAT-and-AUDIO-Mode

I used this to connect to my Raspberry Pi. Still don’t understand why it worked with FlDigi and not with WSJTX but I installed PulseAudio, added the 2 lines in the wiki and it worked. Win 10 just worked! 

Bob Thornton

DG9BFC, 2018-03-18:
or to sum it up ... line out is nearly useless for pc work since we have usb audio and cat ... only usefull thing for line out is if you wanna connect an active speaker or similar thingy to the mchf ... all pc things can be better done via usb audio

WM0I, 2018-03-18:
Nothing is working. I loaded windows 10 USB drivers for the audio.. I get a high pitched whistle and the McHF clone locks up and I have to pull the power connector off to shut it down. There are so many things that need to be fixed it isn't worth messing with anymore. 

Thanks for trying. But I am out of it now. Going to sell it and get a regular transceiver.

73
Richard WM0I

G3WKW, 2018-03-18:
Is that when you transmit? Perhaps RF into the cable. I had that at first but tidied my cables and all was fine. Try at 0.5 watts or long press of Tune to kill RF.

Bob Thornton

WM0I, 2018-03-18:
no, on RX
Richard

DG9BFC, 2018-03-19:
hmmm on rx the signal comes out of line connector into your soundcard ... your soundcard has to process that audio and sends it then to speaker ... that may take a fraction of a second (depending if you additionally use an audio filtering software like dsp.filter from makito mori or a similar filter soft it may take even longer)
so when you compare the audio from internal speaker and line out ... you should run your active (pc) speaker direct from the line out ... then no timing issue should occur ...
i have also a (slight) echo when using my external dsp filter with any rig ... reason is that there is some "speak trough" (unfiltered audio reaches the power amp with a very low strength before the processed audio with normal strength reaches the speaker amp) ... but that is not a problem of the rig used but a problem of the external filter ....
the question is ... do you really hear an echo on external sound??? (means your hear it twice on that one output) ... or do you just hear a timing difference between audio from internal speaker and external sound from pc??

closed 2018-03-20 by WM0I