df8oe/UHSDR #1489: new DSP virtual keypad does not work on small resolution LCDs

df8oe, 2018-04-05:
Your firmware version: 2.9.21
Your bootloader version: 4.0.0

### Hardware 
 * UI Board: mcHF 0.4
 * RF Board: mcHF 0.4

### Describe the issue:

Touching on DSP box gets the virtual keypad - but no key can be selected by touch there. On 480x320 LCD everything works as expected.

 * Display resolution: 320x240
 * Display connected via: Parallel
 * Touchscreen: Yes
 
73
Andreas

sp9bsl, 2018-04-05:
Hi Andreas,
tested this yesterday intensively on 320x240 and it worked. Will look at it.

73 Slawek

sp9bsl, 2018-04-05:
Andreas,
One question:
Do pressing DSP button (G2) changes the "pressed" button on displayed keyboard?

df8oe, 2018-04-05:
Pressing G2 does not work anymore (no reaction). Neither in standard mode or in mode where keypad is shown.

sp9bsl, 2018-04-05:
Ok, thank you this is the default mask problem (disabling the particular dsp function feature). Try to long press the displayed key (keys) to check if it will reenable functions and start to work.

df8oe, 2018-04-05:
Now its getting confusing..... After touching long into "NR" it worked. NR wa enabled and key background of NR switches to black. But when I tap on another button (example "AUTO NOTCH") short again nothing happens. When I tap long into this field knob border changes (like button pressed or not) but it does not change to black and/function is not anabled. When I tap long into NR+NOTCH background changes to orange, function is not enabled and background of NR again stay black (and NR is working). Short tap does not work at all. When background of NR+NOTCH is orange I can cycle using G2 to NR+NOTCH - but only if I tap it to "orange" before. If it has normal background I can only toggle between DSP OFF and NR using G2. Behaviour on OVI40 with 480x320 works like I expect it, mcHF with small LCD is confusing....

sp9bsl, 2018-04-05:
Ok, will check this afternoon. Maybe somethig missed with my tests, sorry.

df8oe, 2018-04-05:
No problem. Maybe you have entered a configuration where it is working. It is impossible to test all possible combinations. That's the proof of community working: normally issues are identified very fast and can be worked out just in time :)

sp9bsl, 2018-04-05:
I think I did pull to early... Anyway, have to add some manual to wiki with state of virtual buttons: normal/pressed and disabled, it is confusing.

df8oe, 2018-04-05:
No I do not think it is too early. On OVI40 it works flawlessly. There is a small part which is not functional (only on mcHF) and I am sure this can be fixed soon. "Bleeding edge" means it *may* contain some bugs (mostly not important ones which make the firmware unusable). It is our standard way of working. Only a programmer who does not create any code does not make any mistakes...

df8oe, 2018-04-09:
I have wait some time if someone else got any issue. It seems not - here ist working all as expected now. Good work!

closed 2018-04-09 by df8oe

sp9bsl, 2018-04-09:
So, I will prepare new virtual keypads in the near future.

df8oe, 2018-04-09:
Highly appreciated! My favorites:

1. filters
2. modes (included digital modes)
3. bands

73
Andreas