df8oe/UHSDR #1494: Saving Goertzel block size

K4LXY, 2018-04-08:
# UHSDR Firmare Defect Issue (Bug) Template 1.0

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Your Issue Data goes here:

Your firmware version: 2.9.24

### Describe the issue:

The block size for Goertzel (cw decoding) doesn't save. It returns to the default when the rig is powered off.  This has been true on all firmware versions that I have tried.  Not a significant problem, but is there any way to allow changes to be saved?  Thanks.

DD4WH, 2018-04-08:
Hi!
thats true, we never implemented saving of the Goertzel block size. It was intended as a temporary adjustable setting for testing of the CW decoder. It seems we forgot about still having it in the menu ;-).

Did you recognize an optimal setting for the Goertzel block size which increases reliability of the CW decoder? If yes, please report, so we can use that block size as the default. Or do you feel the need to adjust Goertzel block size for different CW signal SNRs?

73 Frank DD4WH

K4LXY, 2018-04-08:
Thanks Frank
96 works very well for me - speeds from about. 16wpm to 22 wpm anyway.  I’ve been setting it there and leaving it, and it does a good job.
73
Howard K4LXY

> Hi!
> thats true, we never implemented saving of the Goertzel block size. It was intended as a temporary adjustable setting for testing of the CW decoder. It seems we forgot about still having it in [...]

Howard Zehr
Photo galleries:  
www.howardzehr.com
https://www.facebook.com/HowardZehrPhoto

K4LXY, 2018-04-10:
However, so far it seems that 32000 may work better with stronger signals.  I’ll experiment more.
Howard K4LXY

> 
> threshold at 32000 is way too high for signals close to the noise borde
> 15 to 20 000 (higher if you add the peakfilter!!)

Howard Zehr
Photo galleries:  
www.howardzehr.com
https://www.facebook.com/HowardZehrPhoto

K4LXY, 2018-04-23:
Is there a simple fix that would allow the changes I make to be saved?

DG9BFC, 2018-04-27:
no .... the programmers made them not saveable ... only way is tell programmers the values that you found that work best ... then they can set that as the default value

K4LXY, 2018-04-27:
Thanks
I’ll suggest 15,000 and 96
Howard K4LXY

> 
> no .... the programmers made them not saveable ... only way is tell programmers the values that you found that work best ... then they can set that as the default value

Howard Zehr
Photo galleries:  
www.howardzehr.com
https://www.facebook.com/HowardZehrPhoto

df8oe, 2018-06-26:
fixed.

closed 2018-06-26 by df8oe