df8oe/UHSDR #1495: When band is changed via CAT, "band label" does not change

KX5SP, 2018-04-08:
# UHSDR Firmare Defect Issue (Bug) Template 1.0

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Your Issue Data goes here:

Your firmware version: 2.9.24
Your bootloader version: 3.4.2

### Describe the issue:  When band is changed by CAT command the "band label" (small font below right side of S meter) does not change appropriately, even though the indicated frequency is correct. Example - right now the freq indicates 18.100 but the band indicates "10m."

Other settings you consider as relevant:

 * ...

G3WKW, 2018-04-08:
Sometimes annoying but that is exactly how it is documented so it is not a defect.

https://github.com/df8oe/UHSDR/wiki/USB-CAT-and-AUDIO-Mode

db4ple, 2018-04-09:
Hi,

if you enable the "CAT Running In Sandbox" in the configuration menu, you'll see that the name of the shown band label switches to "CAT" once you use CAT to change frequencies.

DG9BFC, 2018-04-12:
maybe operating error should be changed to operator error?!? :-)

closed 2018-04-20 by df8oe