df8oe/UHSDR #1496: Keyboard CW transmission? (enhancement)

Haasoperator, 2018-04-10:
I was wondering if any progress has been made on allowing CW to be transmitted via USB keyboard, now that RTTY and BPSK have been enabled. Mainly thinking of mcHF use.

WD8BXS, 2018-12-27:
This would be a great enhancement, several people have expressed interest to me about it.

WD8BXS, 2019-01-08:
Is this going to be possible??

m-chichikalov, 2019-01-08:
Sure, sooner or later, I can take a look into this a bit later.  73!

m-chichikalov, 2019-01-18:
@WD8BXS , Hi,
Hi, have you ever seen that when you are typing really fast (kind of fast ;), some characters doubles?  For example, if quickly press "qwe" on the screen it appears as -> "qwwe" and will transmit as on the screen.

Just want to know, maybe this is the issue with my keyboard.
73!

WD8BXS, 2019-01-18:
Hi,No I have not seen this issue.I will experiment.Good news on the CW..Thanks for all of your efforts.
Chuck

m-chichikalov, 2019-01-19:
@WD8BXS , it should be the same on your side, theoretically. Could you try, please, in RTTY or BPSK press 'w' and by holding it press 'e' -> should appear as 'wwe'  No?

@db4ple , @df8oe  could you guys try also, thanks.

m-chichikalov, 2019-01-19:
[image]

Ok, there are two ways to handle it, the first is the easiest one, the second one is the right one... 
1 - send requests to keyboard more really, 
2 - dedicate module to handle keyboards requests, and analyze the entire array...

maybe another?

m-chichikalov, 2019-01-19:
**FIXED**

db4ple, 2019-01-19:
Sorry, yes, I mixed it up with CW entry for PSK.

WD8BXS, 2019-01-19:
Yes it is true, but I don't type that fast hihi.

So as I understand it, the cw from keyboard is NOT quite ready yet??

TNX, 73, Chuck WD8BXS

m-chichikalov, 2019-01-19:
> Yes it is true, but I don't type that fast hihi.
> 
> So as I understand it, the cw from keyboard is NOT quite ready yet??
> 
> TNX, 73, Chuck WD8BXS

You are right, it's not pushed to the official build. It's working, but because everything very coupled together, some changes in one place broke some in others.... 

It's working already but needs some fix for RTTY. What do you using, I can compile for you to try it on and give me feedback, it's also very important. 

I added:
- there are three buttons on the UI, so the context of them could be transmitted by pressing F5,F6 and F7 respectively.  
- You can clean the buffer by pressing ESC on the keyboard.
- Fixed issue with pressing multi buttons on the keyboard.
- The context of those "Three buttons" could be written from the keyboard, not only from paddles.

73! Max RV9YW

WD8BXS, 2019-01-19:
Sound like it is coming along.
I use the MCHF
A friend is interested in the CW from Keyboard on his MCHF.

TNX, Chuck

m-chichikalov, 2019-01-23:
During testing, I also found that sometimes my keyboard not recognized by mcHF. So, I usually fixed it by rebooting TRX. 

> m-chichikalov
> 
> Yes it does work with my wired USB keyboard, but now my Rii Mini X1 Wireless keyboard does not now!!
> The F1 and F2 buttons do indeed put it into transmit and receive, but no letters work.
> Thanks,
> Chuck

Hi Chuck,
During testing, I also found that sometimes my keyboard not recognized by mcHF. So, I usually fixed it by rebooting TRX. 

What CW mode have you used? It would work only with the IAM mode.

WD8BXS, 2019-01-23:
I tried the reboot, but to no avail.  as I said it must see the wireless keyboard as the F1 and F2 work.

I am using IAM mode.

WD8BXS, 2019-01-25:
Testing tonight!! it works but only for a minute or so. I can type in only a few words then it stops, and stays keyed. when I get it unkeyed, the keyboard stops working.

Very good progress though.

I know you will get it going ..

Thanks,
 Chuck WD8BXS

WD8BXS, 2019-01-25:
I just discovered that on 5 watts it works great.. maybe some RF getting into it on 10 watts ??

Chuck

m-chichikalov, 2019-01-28:
> I just discovered that on 5 watts it works great.. maybe some RF getting into it on 10 watts ??
> 
> Chuck

Hi Chuck, sure... High level of vswr should not help, yes maybe Your antenna is fine-tuned. Don't know, but it's a hardware issue that keyboard stop working after 10Wt, It's up to you how to fix it, sorry We cannot help you with that.

ei8hvb, 2019-03-10:
Can I ask pls, is there a list of USB keyboards know to work with MCHF, cabled or bluetooth.

73
Harry

df8oe, 2019-03-10:
Such a list does not exist (and cannot exist because of the huge amount of different keyboards). All standard keyboards (which are no "multiple USB keyboards") should work.

m-chichikalov, 2019-03-14:
At least We can start creating such a list...
@ei8hvb  The Dell SK-8115 works fine.

ok1cdj, 2019-07-08:
I made page in wiki, you can add your keyboard. Actually i am looking for small keyboard which will works.
https://github.com/df8oe/UHSDR/wiki/Compatible-USB-keyboards