df8oe/UHSDR #1498: Can't get into default settings

WD8BXS, 2018-04-14:
Your firmware version: latest
Your bootloader version: latest

### Hardware 
 * UI Board: mcHF 7
 * RF Board: mcHF 7

### Describe the issue:

I can't seem to load the default values. When I press F1 F3 F5 it just goes to the blue test screen.

Also when I touch  he screen it beeps but doesn't do anything. I have performed the touchscreen calibration .

It will not let me into the touchscreen calibration again, just bluescreen telling me that I am touching the screen.

Also in Blue test screen the touchscreen value do change.

Any ideas??

ase-2017stud, 2018-04-14:
Hi,
post the values for Keys Initial & Keys Current shown when you press nothing. Or simply attach a screenshot to your issue. 

BTW, the word "latest" is not a good reference to version 2.9.25, so please use the numbers. If someone looks at this a month or so later, latest is not the same anymore. Thank you!

df8oe, 2018-04-14:
Refactored button test impacts loading of default settings and touchscreen calibration...

db4ple, 2018-04-14:
Hi Andreas,
can you reproduce the problem?

73
Danilo

WD8BXS, 2018-04-14:
Andreas, does this meant there is an issue?? I saw that statement on the last daily build. Don't understand it, but I was it hihi

WD8BXS, 2018-04-14:
Here is the pic you requested

df8oe, 2018-04-14:
Yes. I can reproduce.

df8oe, 2018-04-14:
fixed now.

closed 2018-04-14 by df8oe