df8oe/UHSDR #1499: Receive glitch

WD8BXS, 2018-04-14:
Your firmware version:  D2.9.24
Your bootloader version: 4.0.0

### Hardware 
 * UI Board: mcHF 0.x / OVI40 1.x
 * RF Board: mcHF 0.x / OVI40 1.x

### Describe the issue:
Different radio,

I always have the peak in the center of the screen. Not on my other one.

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

You have audio related problems:
 * DEMODULATION MODE: LSB/...
 * Filter:  2K7 LPF
 * Noise blanker setting: 0
 * Noise reduction: ON/OFF

You have display (also fill if audio problem) related problems:
 * Display resolution: 320x240 
 * Display connected via: Parallel
 * Touchscreen: Yes
 * Magnify: 1x
 * Spectrum: Scope
 
[image]

df8oe, 2018-04-14:
I do have this glitch on most of my mcHFs. Quite normal - no issue.

WD8BXS, 2018-04-14:
Strange, my other one here does not have it.
Chuck

closed 2018-04-14 by WD8BXS