df8oe/UHSDR #1503: Error codes

WD8BXS, 2018-04-14:
Upon power up of a ver 6 I am getting :
" swr bridge resistor mod NOT completed"

What is this referring to please ??
Thanks,
Chuck

WD8BXS, 2018-04-15:
I read through that document. It still doesn't tell me about the error that I asked for. I have never done the SWR mod on any of the radios I have built. And have not seen this error. I suppose it means there is an issue in the swr circuit.

df8oe, 2018-04-15:
Beleave us,. You _must do the SWR mod_ otherwise readings are very unlinear / unaccurate. The detection of the mod maybe not much reliable. The mod is in theory not neccessary **if all GPIOs would have same input impedance**. In theory. But in practice **impedances are different**. So you do not symmerical measuring of FWD and REV. Clint KA7OEI has investigated this and as an answer output impedance of SWR bridge is reduced so that hopefully different impedances on GPIO side do not impact much because they are much higher than new output impedance of SWR bridge.

**Do the mod** and be happy.

vy 73
Andreas

df8oe, 2018-04-15:
Additional:
I can only speak about RF v0.4. I do not know if Chris has implemented changings into newer versions - please take a look at the schematics.

WD8BXS, 2018-04-15:
I believe you, but what does the error message have to do with the mod? I have never done the mod I have never got this error message. I just want to know specifically what the error message refers to. There has to be something in mind for this error message to come up. What is the radio software seeing to trigger this message?

df8oe, 2018-04-15:
Hi Chuck,

if the impedances which are connected to the ADCs for FWD and REF are too high the voltages at this pins are differing. If such a differing is detected the error message is thrown. Normally this means that the SWR mod is not done otherwise there would not be a difference (because it is minimized by reducing the connected impedance). Maybe you are a very lucky man and all your STM32's do have nearly similar impedances on both ADCs so that the mod is not mandatory. But this is not normal case. I have built ~20 mcHFs and to all of them the SWR mod must be applied.

db4ple, 2018-04-15:
Hi Chuck,
I already explained this a few comments above: The firmware measures the voltage level on the two SWR bridge outputs via ADC. If the voltage level on both, REV and FWD, is above a certain predefined threshold, code assumes that you did not do the SWR mod. The detection code DOES NOT care if there is a imbalance between both:

ui_driver.c line 5720 (roughly):
```
UiDriver_StartupScreen_LogIfProblem(

(HAL_ADC_GetValue(&hadc2) > MAX_VSWR_MOD_VALUE)
&& (HAL_ADC_GetValue(&hadc3) > MAX_VSWR_MOD_VALUE),

"SWR Bridge resistor mod NOT completed!");
```
 
As Andreas said, this mod (and the subsequently written problem detection code) was done by Clint, KA7OEI, more than two years ago when the mcHF 0.4 was out.

WD8BXS, 2018-04-15:
AH, very well, I will check into that. I guess I am a very lucky man, over 21 built and never had this issue before, and I have never moved the coil wire.

Thank you for the information, I will close now!

closed 2018-04-15 by WD8BXS