df8oe/UHSDR #1506: Touchscreen calibration routine broken in 2.9.27

df8oe, 2018-04-18:
At .27 version touchscreen calibration does no longer work. Crosses are confirmed "automatically" without pressing TC and calibration goes completely wrong because of this.

WD8BXS, 2018-04-19:
I wonder if this is related to my issue? 
I touch my screen, I hear the beep, but nothing changes.
When I am in test screen, I touch the screen and hear the beep but screen reports no coordinates.  Raw data changes but x,y says zero

Same issue or no??

Thanks,
 Chuck WD8BXS, 73

db4ple, 2018-04-19:
Hi Chuck,
no, not related. Not directly at least. AFAIR your problem was older then 2.9.25, so it cannot be caused by this specific issue introduced in 2.9.25
But if the raw data changes and the coordinates are both zero -> this indicates are completely "off bounds" calibration . To fix: Rerun the calibration using 2.9.28 or later.

73
Danilo

WD8BXS, 2018-04-19:
Thanks, That did it!!

closed 2018-04-20 by df8oe