df8oe/UHSDR #1508: Question: SWR

Groc-1, 2018-04-22:
Newbie to mcHF

1> Back in late 2016, the answer to SWR requests was ">=5w needed in the mcHF for SWR detection",
Is this still the case?
2> Does the V.06+ "fold back" the power in the presence of high SWR?
3> If not, is there any protection for the finals

db4ple, 2018-04-22:
So the conclusion is: you can kill something with high SWR but most likely not the finals. Which does not make it better, to be honest... Maybe we should indeed add some power reduction. But we would need some data for that to make sense.

Groc-1, 2018-04-22:
If implemented, suggest foldback to start at 3:1 SWR.

df8oe, 2018-04-28:
If you are using a mag loop you can pretty tune during RX. If you are approaching best match RX level increases significantly, When you tune to "best RX" your SWR should be better than 3:1

K4LXY, 2018-04-28:
Mag loops have such low impedance that any mismatch can result in significant loss.  They also have extremely sharp tuning and thus a very narrow bandwidth.  I'd recommend some kind external SWR indicator to be able to tweak the match for maximum efficiency.
Howard K4LXY

Groc-1, 2018-04-28:
Using a You-Kits FG-01A to pre-tune..
Graphically indicates frequency, SWR and impedance ..
Small and very effective..

73

Dave G.    KK7SS
Richland, WA.

K4LXY, 2018-04-28:
Yes - that works!
Howard K4LXY

> 
> Using a You-Kits FG-01A to pre-tune..
> Graphically indicates frequency, SWR and impedance ..
> Small and very effective..
> 
> 73
> 
> Dave G. KK7SS
> Richland, WA.
> 
> 
> > Mag loops have such low impedance that any mismatch can result in
> > significant loss. They also have extremely sharp tuning and thus a very
> > narrow bandwidth. I'd recommend some kind external SWR indicator to be able
> > to tweak the match for maximum efficiency.
> > Howard K4LXY
> >
> > <https://github.com/df8oe/UHSDR/issues/1508#issuecomment-385184115>, or mute
> > the thread

Howard Zehr
Photo galleries:  
www.howardzehr.com
https://www.facebook.com/HowardZehrPhoto

df8oe, 2018-04-29:
If you are satisfied with the informations: please close this issue. Thank you!

Groc-1, 2018-04-29:
Good info.. thanks

closed 2018-04-29 by Groc-1