df8oe/UHSDR #1509: DTR & RTS signalas and CW

ur7hfo, 2018-04-24:
I tried a fragment of your code reading the RTS and DTR signals.
He did not earn from me correctly.
Earned the following option

case CDC_SET_CONTROL_LINE_STATE:
      // we get wValue here as buffer  (rts & dtr)
      // wValue = *(uint16_t *)&(pbuf[0]);            <<=== ERROR !
      wValue = pbuf[2];
      cdcvcp_ctrllines.dtr = (wValue & 0x01)?1:0;
      cdcvcp_ctrllines.rts = (wValue & 0x02)?1:0; 
    break;

Taken from - https://github.com/df8oe/UHSDR/blob/active-devel/mchf-eclipse/drivers/usb/app/usbd_cdc_if.c

db4ple, 2018-04-24:
Hi,
I don't exactly know what you want to tell us.  Please explain. What kind of error have you found?
DTR and RTS are bits 0 and 1 in the first 16 bit of the payload (== pbuf[0] and pbuf[1]). This follows the CDC ACM standard. And the UHSDR is known to work with a number of tools exactly with this code like UCXlog and more, see the wiki: https://github.com/df8oe/UHSDR/wiki/USB-CAT-and-AUDIO-Mode#programs-used-to-receivetransmit-cw
 

Please give us more details about your setup (see the issue form). Also relevant here is, how did you test that your proposal works?  Is any of the list programs working for you?  
BTW, I just tried N1MM+ with RTS for PTT (need to enable that in UHSDR configuration menu) and DTR for CW key. Current UHSDR firmware 2.9.28 works without problems for me under Windows 10 with this problem.

ur7hfo, 2018-04-24:
> DTR and RTS are bits 0 and 1 in the first 16 bit of the payload (== pbuf[0] and pbuf[1]). 

The matter is that at me, rts & dtr it is transferred in the third byte. And this is confirmed ..
Here is a quote:

In the SET_CONTROL_LINE_STATE request (22h), the host tells the device how to set the RS-232 control signals RTS and DTR. The host sends the control-line states in the **third byte** of the Setup transaction. Bit 0 is the state of DTR, and bit 1 is the state of RTS. Device firmware detects the request, accepts the data, and implements any changes to the bits. The IN transaction is the Status stage. The device indicates success by returning a ZLP.

Here is the link to the article :  http://janaxelson.com/usb_virtual_com_port.htm

ur7hfo, 2018-04-25:
Good afternoon. Thanks for your reply.
I use in this case N1MM+

Maybe I will need to update the USB NAL driver, because in my version the line of code specified by you looks like this:

((USBD_CDC_ItfTypeDef *)pdev->pUserData)->Control(req->bRequest,
                                                        (uint8_t*)req,
                                                        0);

And in your version of the driver, the code looks like

 ((USBD_CDC_ItfTypeDef *)pdev->pUserData)->Control(req->bRequest,
                                                        (uint8_t*)&req->wValue,
                                                        0);

closed 2018-04-25 by ur7hfo

db4ple, 2018-04-25:
Yes. I don't know which code you have been looking at, but it would have been nice to tell us that you were not using the latest firmware source code and that you did not try to use the official latest builds as reference.

ur7hfo, 2018-04-26:
Yes, I do not use the last of your assemblies, and I do not use a transceiver
mcHF QRP transceiver.
I needed in my small project on stm32, to use the signals of rts and dt from yusb. I did not know how to do this until I found your project on the Internet.
Thank you for attention.