df8oe/UHSDR #1511: Power adjustment FULL power dows not work

la2fda, 2018-04-27:
## Your Issue Data goes here:
Your firmware version: 2.9.29
Your bootloader version: 4.0.0

### Hardware 
 * UI Board: mcHF 0.5 
 * RF Board: mcHF 0.5

### Describe the issue:
Under initial adjustments:
5w PWR Adjust works fine.
When selecting Full PWR Adjust:
* The figure for the selected band is white
* When pressing TUNE it is going yellow, and is not possible to adjust.

df8oe, 2018-04-28:
This normally indicates you are not transmitting with FULL PWR when trying to adjust.

TUNE power cvan be set individually and different from normal TX power in menu. You must either set TUNE PWR to "FULL" or to "as TX PWR" while TX PWR is set to FULL...

la2fda, 2018-04-28:
This does not behave like normal.
The value for the selected band is white ( it is not white before i have the correct Power (FULL) )
When I Press Tune i expect it to be possible to change the value ( Like it does on 5w PWR adjust )
What happen is: Instead it is turning yellow (Like the values for all the other band)l

db4ple, 2018-04-28:
As Andreas said, check your tune power settings if it is set to "5W", it is clear way Full Power Adjust doesn't work (Tune switches to 5W).

la2fda, 2018-04-28:
Ahh, Tune Power switches. I know, but had forgot.
Sorry. Case closed !
Thank you !

closed 2018-04-28 by la2fda