df8oe/UHSDR #1512: [improvement] Is IQ recording possible?

dmitryelj, 2018-04-30:
Hi all. 

I'm interested in SWL and signals analyse, and it will be cool to have this feature in the portable radio. I think, there are no transceivers in the market, that have IQ recording possibility.

So, 2 questions: 
- is it possible to connect micro SD-card to the board? Maybe using GPIO directly, or via USB? 
- if yes, can IQ-recording be added in the future?  

Thanks.

DD4WH, 2018-04-30:
There are several SDR radios on the market that are able to do IQ recording/playback. For example, the Perseus SDR can do that since 2008. Even unattended timed recordings can be done using other small helper programs with the Perseus.
Only my personal opinion:
It is very unlikely that we will implement IQ recording on SD card for OVI 40/mcHF:
* recording to SD card is very hard to achieve. It needs memory of at least 200MB to buffer the IQ files when the SD card is not yet ready to write. Modern SD cards can have write latencies of several hundred microseconds or even up to seconds.
Recording to flash memory: 
* why not. With an additional flash memory, recording and playback should be possible. But that also needs a hardware extension (the flash) and heavy programming.
So, maybe in the far future, but unlikely.
All the best 73s,
Frank

dmitryelj, 2018-04-30:
Thanks. I was talking about portable SDRs, not about Perseus, of course.

If its impossible in this hardware revision, we can close the issue for now.

closed 2018-04-30 by dmitryelj

DD4WH, 2018-04-30:
OK, you are right, maybe Reuter Pocket will have that ability in the near future. At the moment only audio can be recorded, not IQ yet.

df8oe, 2018-04-30:
Yes - it is already possible. You only have to choose "DIQ" as input. If you do so USB audio input AND output is I/Q. Now connect a laptop via USB to mcHF and record :) I do not think it makes any sense to add possibility for "standalone recording"....