df8oe/UHSDR #1516: Strange glitch

WD8BXS, 2018-05-05:
## Your Issue Data goes here:

Your firmware version:2.9.29
Your bootloader version: 4.0

### Hardware 
 * UI Board: mcHF 7.0
 * RF Board: mcHF 7.0

This radio keeps booting up in load defaults (red screen)  mode.
No buttons are pushed, I have proven that with the blue test screen.
Seems that if I leave the sides off it boots Up normally.

Not sure where to start looking.

Thanks,,
Chuck WD8BXS

df8oe, 2018-05-05:
Check all tracks from F-buttons to MCU. Maybe there are short circuits or connections to the case if it is closed. This sounds like a hardware issue.

73
Andreas

WD8BXS, 2018-05-06:
Very interesting problem, I have found out that if you just press power and let it go ahead and boot, it will continue to boot flawlessly.

BUT if you let it sit for about 5 minutes, then it will boot into the “Load Default” screen.

Must be a weak component somewhere that cools off or something.

Wonder where to look?

De WD8BXS, Chuck

WD8BXS, 2018-05-08:
I think it is because I used 100nf caps on the encoder headers, I will change to 10 nf and report back.

Groc-1, 2018-05-10:
Following this with interest :-)