df8oe/UHSDR #1522: CW Filters - enhancement

HB9FIH, 2018-05-17:
No others is relevant, i'ts a view for CW Filters.

If I set in CW Mode Menu - my CW Offset to 700 then for me is 700 Hz the middle and the way I work CW...

The 4 CW Filters now could all be centered based at the offset value with different bandwith:
eg: 100 / 200 / 300 / 500

df8oe, 2018-06-09:
At the recent stage there are only 4 filters available with center frequencies of 500Hz and 750Hz. We will not apply more filters because they will not fit in small MCU of mcHF. On OVI40 there will be "free adjustable (runtime calculated) filters in the future - but not on mcHF bwcause of too low horse power.

HB9FIH, 2018-06-09:
Tnx ok.
I have a workaround with Audio Filter which works v well. (NSCAF from uk qrp group). I only have a small box more.
It was a suggestion...ur firmware is great!
Gruss Erich

closed 2018-06-26 by df8oe