df8oe/UHSDR #1525: Will there be more firmware updates? Just a question.

WM0I, 2018-05-28:
# UHSDR Firmare Defect Issue (Bug) Template 1.0

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Your Issue Data goes here:

Your firmware version:
Your bootloader version: 

### Hardware 
 * UI Board: mcHF 0.x / OVI40 1.x
 * RF Board: mcHF 0.x / OVI40 1.x

### Describe the issue:

Place tell us what does not work and how others can reproduce the issue (list related settings, steps to do, etc.):

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

You have audio related problems:
 * DEMODULATION MODE: LSB/...
 * Filter:  2K7 LPF
 * Noise blanker setting: 0
 * Noise reduction: ON/OFF

You have display (also fill if audio problem) related problems:
 * Display resolution: 320x240 / 480x320
 * Display connected via: Parallel / SPI
 * Touchscreen: Yes/No
 * Magnify: 1x / 32x
 * Spectrum: Waterfall/Scope/Dual
 
Other settings you consider as relevant:

 * ...

sp9bsl, 2018-05-28:
Hi Richard,
yes!! Please be patient, we are working...

73 Slawek

K4LXY, 2018-05-28:
Is there any chance that in a future update you could allow operator changes to the decoder settings to be saved?
73 Howard K4LXY

> 
> Hi Richard,
> yes!! Please be patient, we are working...
> 
> 73 Slawek

Howard Zehr
Photo galleries:  
www.howardzehr.com <http://www.howardzehr.com/>
https://www.facebook.com/HowardZehrPhoto

df8oe, 2018-05-28:
Please:

- if you think you have found an issue: tell us how to reproduce.

- if you want an enhancement: please describe what kind of enhancement (working scenario).

Thank you!

vy 73
Andreas

K4LXY, 2018-05-28:
I submitted a comment earlier (1494) on my question about the decoder.
Thanks
Howard K4LXY

> 
> Please:
> 
> if you think you have found an issue: tell us how to reproduce.
> 
> if you want an enhancement: please describe what kind of enhancement (working scenario).
> 
> Thank you!
> 
> vy 73
> Andreas

Howard Zehr
Photo galleries:  
www.howardzehr.com <http://www.howardzehr.com/>
https://www.facebook.com/HowardZehrPhoto

DD4WH, 2018-05-29:
Hi guys,
yes, there will be more firmware updates. There is some silence at the moment due to three reasons:

1.) At the moment its too hot here in Central Europe to work on these things (hottest month of May ever) ;-).

2.) At the moment work is going on for the new RF PCB hardware, theoretical considerations on a complete filter rebuild and new ways of transmit signal production. But its much too early to talk about this.

3.) we are going to present the status and our new plans on hardware and software at the SDR Academy at the HAMRADIO exhibition in Friedrichshafen this weekend. We also hope for new ideas and impulses to integrate into our software. 
http://2018.sdra.io/pages/programme.html

So, to summarize: yes, we are busy, but at the moment you will not notice it on github :-).

73 Frank

mm0mhp, 2018-05-31:
Thanks Frank

For those who cannot attend do you know if the presentation will be live streamed?

73 Harry

DD4WH, 2018-05-31:
probably no live stream.
But last years´ talks are on youtube now, I guess the same will happen with the SDR academy 2018 talks.
73 Frank

closed 2018-06-09 by df8oe