df8oe/UHSDR #1527: wm8731 v TLV320AIC23

sv1it, 2018-06-01:
# UHSDR Firmare Defect Issue (Bug) Template 1.0

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Your Issue Data goes here:

Your firmware version:
Your bootloader version: 

### Hardware 
 * UI Board: mcHF 0.x / OVI40 1.x
 * RF Board: mcHF 0.x / OVI40 1.x

### Describe the issue:

Place tell us what does not work and how others can reproduce the issue (list related settings, steps to do, etc.):

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

You have audio related problems:
 * DEMODULATION MODE: LSB/...
 * Filter:  2K7 LPF
 * Noise blanker setting: 0
 * Noise reduction: ON/OFF

You have display (also fill if audio problem) related problems:
 * Display resolution: 320x240 / 480x320
 * Display connected via: Parallel / SPI
 * Touchscreen: Yes/No
 * Magnify: 1x / 32x
 * Spectrum: Waterfall/Scope/Dual
 
Other settings you consider as relevant:

 * ...

db4ple, 2018-06-01:
Hi,
this might not be the right place to ask, this question is better placed in the Yahoo NG or the I40 Forum since you will reach a much larger audience there.  

I at least have no knowledge if there is any difference or not. Does the firmware report a missing audio codec (boot screen  and/or system info menu)?

73
Danilo

sv1it, 2018-06-01:
Hi Danilo,

No everything seems tobe ok but….

Thanks anyway and 73

Kostas

df8oe, 2018-06-04:
These two codecs are 100% hardware and software compatible. I can confirm this because I used hundreds of them in OVI40. so it is definitely NO firmware issue.

vy 73
Andreas

sv1it, 2018-06-08:
thanks Andreas

73
Kostas

closed 2018-06-08 by sv1it