df8oe/UHSDR #1532: CW Keyer - Problem sending Dahs

devmrs, 2018-06-12:
# UHSDR Firmare Defect Issue (Bug) Template 1.0

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Your Issue Data goes here:

Your firmware version: S2.9.34
Your bootloader version: 4.0.0

### Hardware 
 * UI Board: mcHF 0.6
 * RF Board: mcHF 0.6

### Describe the issue:
I have a Shengda SD010HF that came with firmware 2.2.9. The keyer works perfectly in that version. Using S2.9.34 (and many prior versions), pressing the DAH side of the key results in "DI-DAH". I discovered that if I first send a letter beginning with DI-DAH (A or W or J) the keyer works correctly until the RX->TX timeout which resets the keyer. I upped my RX->TX time to 200ms. The keyer now works correctly after first sending A or W or J (until RX->TX timeout).
[image]

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

 * ...

df8oe, 2018-06-13:
I have just tested: All works as expected @2.9.34. I cannot confirm. Pressing dah always gives a dah and never a di-dah.

df8oe, 2018-06-13:
When looking at your firmware revision number I notice that it is not an "official build". Oficial builds are starting with a "D" - not with a "S". Maybe that is the problem?

devmrs, 2018-06-13:
I downloaded my S2.9.34 from: https://www.amateurfunk-sulingen.de/uhsdr/archive/firmware/2.9.34/mchf/MCU_512KB/fw-mchf.bin

Apparently I don't know where to download D2.9.34

devmrs, 2018-06-13:
FYI: I don't have enough flash memory in my radio to download https://www.amateurfunk-sulingen.de/uhsdr/archive/firmware/2.9.34/mchf/fw-mchf.bin. Is that my problem? Does "S" stand for the Smaller version of the firmware located in the MCU_512KB/ folder?

devmrs, 2018-06-13:
Versions D2.5.73 and lower all seem to have the keyer working correctly (on my radio).
Starting with D2.5.74 through at least D2.5.80 my radio won't finish booting. The boot process hangs up.
I downloaded D2.6.0 and that version gives me "DI-DAH".
It seems that versions higher than D2.5.73 produce "Di-Dah" when pressing the "Dah" side of the paddle.

df8oe, 2018-06-13:
I repeat: I cannot confirm on any radio I have tested (actually 5 mcHFs, of course all do have 1M of flash). Additionally I never have had used small builds before because I do not have access to mHF using small MCU (and I do not know anybody who owns such a rig). I have tested small build on my mcHF now and can conirm, its version starts with "S..." - but it does not have the issue you described. It is working flawlessly and producing "Dahs" at the beginning (every beginning that I want).

devmrs, 2018-06-13:
Thanks for your testing. It must just be my radio. I'll have to deal with keyer issue on my own. Sending a beginning "W" isn't too bad of a work-around and I can back-level my firmware. You might as well delete this issue as it apparently does not affect any other radios.

df8oe, 2018-06-13:
I leave it open for some time because t would be interesting to see if it is a setting issue or a hardware issue.

To test if it is a setting issue please start using default settings. Then try to key and take a look. If you do not store these settings by long press of MENU button or by switching off radio by switch but by removing power you do not loose yor personal settings.

The only hardware issue I can imagine is a wrong rise time of voltage at dah or dit paddle tracks. First add an additional pullup of 1...3.3K to 3.3V to each line and take a llok and as a second step change capacitors to ground on both lines.

devmrs, 2018-06-13:
I reset everything to default and now keyer works! The only thing I see that is different is that the waterfall was turned off and now it is displayed. I do not see where the option is to turn off the waterfall. I prefer the larger scope without the waterfall.

devmrs, 2018-06-13:
Here are my Display Options:
[image]

df8oe, 2018-06-13:
The option to disable scope or WF is under "Touchscreen via Menu" or (better) by using touchscreen without calling menu...

devmrs, 2018-06-13:
Found it!! Thanks!

closed 2018-06-14 by df8oe