df8oe/UHSDR #1536: SSB filter 4.0khz

ac0ax, 2018-06-24:
# UHSDR Firmare Defect Issue (Bug) Template 1.0

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Your Issue Data goes here:

Your firmware version:2.9.36
Your bootloader version: 4.0.0

### Hardware 
 * UI Board: mcHF .6
 * RF Board: mcHF .6

### Describe the issue:

Place tell us what does not work and how others can reproduce the issue (list related settings, steps to do, etc.): I was playing around with the filter settings and when I set the ssb filter to 4.0k/LPF the waterfall and spectrum scope slowed way down, almost froze. Other filter settings don't seem the have the issue.

df8oe, 2018-06-25:
Many thanks for your report! That is not a bug - it is due to too low horse power of STM32F4. Filters > 3.9KHz do need much more horse power - please use filters <= 3.9KHz for SSB.

ac0ax, 2018-06-25:
What I found strange about it is if I go to 5khz or more it doesn't freeze only on 4khz. I normally use 2.3khz filter but there was a really wide signal and I wanted to see if it would sound different with a wider filter.

closed 2018-06-25 by ac0ax