df8oe/UHSDR #1537: Scope settings in split mode

lilo-chita, 2018-06-26:
# UHSDR Firmare Defect Issue (Bug) Template 1.0

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Your Issue Data goes here:

Your firmware version: 2.9.33
Your bootloader version: 4.0.0

### Hardware 
 * UI Board: mcHF 0.6
 * RF Board: mcHF 0.6

### Describe the issue:

Whatever you set Scope Div. ( MENU_SCOPE_DB_DIVISION) to in Display Menu in split mode (scope + wfall) scope resolution is always 10db/div.

Also Spectrum Filter ( MENU_SPECTRUM_FILTER_STRENGTH) in split mode behaves a bit strange. With the same values in split mode scope is much faster than in normal (scope only) mode.

lilo-chita, 2018-07-03:
Thanks,
scope div is OK now.

What about filter settings?

df8oe, 2018-07-03:
Can you give an exact example what filtersettings you tried to get the effect?

df8oe, 2018-07-03:
OK - I never have had touched that setting since I joined mcHF 3 years ago. This setting is a very old one (I think implemented by Clint). Sorry - I cannot tell you. My setting is "4".

sp9bsl, 2018-07-03:
It is reworked by us when we implemented the dual scope+waterfall view. We have scheduler which calls waterfal **and** scope draw with different timeouts. Both calculate data for the same FFT and filter. The refresh depends on the ratio how often spectrum is drawn to waterfall drawing. It needs rework but is somewhat complicated. We will return to this issue in free time or when the new resolution (800x480) will be implemented - this year I think.

df8oe, 2018-07-03:
Many thanks for this information Slawek. I think this can lowest priotity - and "big resolution" will only work on OVI40 I think.... For all others: use filter "4" and be happy :) I do not think this is an issue. If the starter of this is thinking similar: pse close thos. We will work on refresh rates later.

lilo-chita, 2018-07-03:
It's not critical at all.
Just a note.

closed 2018-07-03 by lilo-chita

df8oe, 2018-07-04:
Many thanks for your report. Good to know, that there may be something buried, which pops up critical many times later (when it can become critical) :) Now we know and will take care about it when we work at this area.