df8oe/UHSDR #1540: Question: (not bug) PSK/RTTY Functionality

Jmelz, 2018-06-27:
How well is PSK / RTTY working? That's vague for a reason - 

I attempted to use PSK31 last night. I thought I had everything set per the wiki, but could not get a decode. When I pressed F1 on a USB keyboard, typed, and then pressed F2, I got the impression that either my text was not being buffered, or worse, maybe not going out. I did not test with another radio yet. I wanted to ask, what indication does a user have that the text is being sent completely? If I type really fast, and hit F2 when done, the radio immediately goes to RX. Maybe the text was cut off because of this?

I'll do more testing, but just curious about this... I believe I have the correct filters in use (from other posts it looks clear), but I wonder if the radio should not be doing better at decoding... CW works very well by contrast.

df8oe, 2018-06-30:
PSK works well, too. But there is no mighty "AFC" function which auto-calibrates frequencies as proper as they are needed for good decoding. Such a routine eats much horse power and RAM - both not available at mcHF.... So you have to fine-tune RX frequency in 1 Hz steps to get best decoding result. It's a bit tricky - but there is no other way.

Jmelz, 2018-06-30:
I must be awful at using it then.  I can't seem to get it working consistently. I'm always using increments of 1... But still poor decode.

Also, still not quite understanding how transmit works... I haven't seen many YouTube vids about this functionality, so I made my own. 

Am I doing something wrong here?

df8oe, 2018-07-26:
We are on the edge for F4 MCU. I do not think we can improve anything else (regarding "autoadjustment" for PSK). And this is the point why it does not work like PC applications...

Regarding "F1": this is not the keyboard but the mcHF button layout we are talking about.

phaethon, 2018-10-01:
@Jmelz Can you describe what issues you have with PSK31 transmit? Receiving part could be improved, but transmit should work well (also, compared to PC).

Jmelz, 2018-10-01:
Last time I messed with it, as I was typing, decided gibberish text interfered with the text I typed.

phaethon, 2018-10-01:
This description does not help much. Do you have repeatable issues with transmit that you can demonstrate? E.g. record with another radio nearby.

Jmelz, 2018-10-01:
I'm not sure how to clarify, but it's like this...

when I switch to psk31 mode, I being typing, and the little tiny space where you expect the text you are typing to be displayed - well, it shows not only my text I'm typing, but received text as well. So, a sentence might look like this:

I type "CQ CQ CQ de W4CKA" and see on the MCHF:

C.3.#>#$>Q CQ C2##ILJ...de3l3.W.ER4CK.3ALA

Does that help?

phaethon, 2018-10-01:
Well, if you type slowly and don't use buffered input, then it just decodes rx in between characters, which is normal. Please, read wiki on buffered vs unbuffered sending.

Jmelz, 2018-10-01:
With a magnetic loop and hours of time, I could only get it to decode maybe 1 out of 20 times. I was using 1 HZ steps. I agree about FT8. But PSK31 on the MCHF is an exercise in frustration.

df8oe, 2018-10-02:
Is your question answered now? If so: please close this issue.

Jmelz, 2018-10-02:
I guess it is.

closed 2018-10-02 by Jmelz