df8oe/UHSDR #1541: -36Khz image poor rejection , only -15db

nizar33, 2018-06-29:
# UHSDR Firmare Defect Issue (Bug) Template 1.0

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Your Issue Data goes here:

Your firmware version:   2.9.41
Your bootloader version: 4.0.0

### Hardware 
 * UI Board: mcHF 0.x / OVI40 1.x
 * RF Board: mcHF 0.x / OVI40 1.x

### Describe the issue:

Place tell us what does not work and how others can reproduce the issue (list related settings, steps to do, etc.):

First thank you very much for the great and wonderfull  work developpement.
Congratulation for the wide sucess. 
We have on audio output a poor -36Khz Image rejection (only -15db rejection)   at -24Khz (-22db)  at -12Khz (-25db) , so we can receive for example FT8 signals at 14110 Khz as well  as at 14074Khz  with only -15db rejection, note that IQ Auto-balancing was set to "ON"  and no image issues seems on Spectrum scope  nor on S-meter  they are both clean , only on audio output signals, othewise reception has a good quality  except this issues,
May be I miss some thing to adjust, please Helpe      
Thank you very much   

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

You have audio related problems:
 * DEMODULATION MODE: LSB/...
 * Filter:  2K7 LPF
 * Noise blanker setting: 0
 * Noise reduction: OFF

You have display (also fill if audio problem) related problems:
 * Display resolution: 320x240 / 480x320
 * Display connected via: Parallel / SPI
 * Touchscreen:No
 * Magnify: 1x
 * Spectrum: Scope
 
Other settings you consider as relevant:

 * ...

df8oe, 2018-06-29:
I cannot confirm. At my mcHFs there is only one "mirror" (it is "at the opposite side of LO" in Xlate distance)  and this one is suppressed >55dB with automatic IQ correction - no other RX mirrors. I think there is a hardware issue at your device.

nizar33, 2018-06-29:
Thank you very much for quick response ,
yes it's a RS928 plus .... very poor really ....

pse if any adjustement proposal or modification ?? may be sound card
Sample Rate software adjustement , if it's possible and How ??

a facebook friend has an original McHF Kit and your newest FW d2.9.42
and he  has approximatly the same audio images issues,  the problem
was not present on spectrum nor on Smeter  indicator , it's only on
decoded audio output, that's may mean it's not a commun hardware
problem,   just a thought ,
we are  many OM's to have this problem
Please if possible helpe
Thank you very much

df8oe, 2018-06-29:
I never have had that on mcHF - never. And we never have had any issue reports about something like this (which are firmware related - of course RX mixer hardware issues can cause this).

Hardware problem in R928Plus is that it uses SI5351 and relies upon LO is always 4 x f(working). But that is not so - we are going a complete different approach on OVI40 where SI5351 is used.

Modification:
Desolder SI5351 and 27MHz TCXO and add SI570. Footprints are present - I have not tested if it will work. But on mcHF and its clones you cannot use SI5351 - it is a complete different approach and only works on a handful firmware releases (where we first add SI5351 support to see if it works in general and then quick switched to our "OVI40 related working way")!

DD4WH, 2018-06-29:
Sorry folks:
There will be NO SUPPORT for cloned radios from my side, not on the hardware and not on the software side.
You all knew in advance what your are buying: a clone.
So you have to live with the fact, that the clone you bought does not work properly.
It would be nice if you tell all your HAM Radio colleagues on the "clone" social networks that the clone you bought is not working.
73 Frank

df8oe, 2018-06-29:
That is my way of thinking, too. No further hints, no further support. My hint ar my lasr post was the only one I have given.

Again: there is **no issue report from any mcHF owner about that** - so it is hardware related and not UHSDR relarwd - so I will close it.

closed 2018-06-29 by df8oe

nizar33, 2018-06-29:
Thank you very much for your promped responses ,
i Made the bad choice to by a clone, it may be due  to not weld the
(mosfets) finals .....
Good Continuation on this great Ham Project .
thanks

df8oe, 2018-06-29:
But:
mcHF *is* a DIY project and there is no way to legally buy a "ready one"! You can turn it like you want: if you do not want to build it you only can buy a "used one" from another OM who offers his self build one...