df8oe/UHSDR #1542: Why is the Red LED Flashing during AM reception?

Groc-1, 2018-07-02:
I notice that the Red LED flashes on peaks when receiving AM transmissions.
Is there a purpose for this?
If not, I cannot find an option to turn it off (excluding the option under CW settings)

## Your Issue Data goes here:

F/W: 2.9.42
B/L:  4.0,0

### Hardware 
 * UI Board: mcHF 0.6
 * RF Board: mcHF 0.6

### Describe the issue:
80 M band, listening to the AM net. RED flashes in sync with carrier peaks on medium to strong signals (S7+).

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

You have audio related problems:
 * DEMODULATION MODE: LSB
 * Filter:  2K7 BPF
 * Noise blanker setting: 0
 * Noise reduction: ON/OFF

You have display (also fill if audio problem) related problems:
 * Display resolution:  480x320
 * Display connected via: Parallel / SPI
 * Touchscreen: Yes
 * Magnify: 4x
 * Spectrum: Scope
 
Other settings you consider as relevant:
DC Supply: Linear PSU, 12.3V

df8oe, 2018-07-02:
Red LED is mostly used for testing/debug purposes. I cannot confirm red LED is flashing in AM so I cannot tell you... There must be some setting different between yours and mine RX.

Groc-1, 2018-07-02:
I'll go through the settings and see what I have set differently from
defaults...

Dave G.    KK7SS
Richland, WA.

lilo-chita, 2018-07-03:
Set CW decoder to OFF

df8oe, 2018-07-26:
If this solves your problem pse close the issue.

Groc-1, 2018-07-26:
That solved the problem... BUT,  setting it to OFF is not permanent; also, I don't see the need to haave the CW LED flash when the mode is NOT set to CW !!  I will close this anyway  :-)

closed 2018-07-26 by Groc-1

df8oe, 2018-07-26:
Of course you are right. But I was missing the feedback if this was the reason for your problem so I have not touched code. Now - as confirmed - I changed it in 2.9.45. Please ALWAYS confirm. There is more than one possible reason for a malfunction and I cannot read your mind ;)

Groc-1, 2018-07-26:
Sorry...  :-(

df8oe, 2018-07-26:
No need for excusing. So it only lasts longer than normal :) Thank you for your reporting!