df8oe/UHSDR #1549: All the reports and help I gave you I get nothing from you?

WM0I, 2018-07-23:
How about supporting what firmware and bootloader you wrote and publish?

"Sorry - no aupport for illegal clones.

73
Andreas"

db4ple, 2018-07-23:
Hi Richard,
we do not support the RS928plus. It is an TRX which does have undocumented significant hardware modifications (unpublished) and uses a modified firmware (unpublished changes, in violation with GPLv3). 

The company responsible for support is the one who did the hardware and software and sold you the device. 

I will not go into the discussion if this is an illegal hardware clone or not, but it is for sure a commerical TRX running the UHSDR software in violation to the terms of the GPLv3. And even if it would not be any of that, no one has an obligation to support anyone.  

I understand your frustration but the device we test the UHSDR mcHF firmware variant against is the mcHF not any of the chinese commercial devices. 

Background: 
I have heard (not seen any of these, so it may not be true) the RS928plus does use the Si5351A clock generator chip where the mcHF uses the Si570 chip. In order to run the Si5351A in the mcHF, one needs to modify the software (which is completely legal under GPLv3, anyone can do this if they follow the GPLv3) and give all users of that software access to the source code (AFAIK they don't do this and with this they violate GPLv3). 

And yes, there is experimental support for the Si5351A in the UHSDR but not for use in the mcHF hardware. Why did the chinese use the Si5351a (I can only guess here): The Si5351a is about 10 Euros or so cheaper so the chinese manufacturer seems to have decided to save some money but this makes this device incompatible with all mcHF firmware builds, not so good for the users. And there may be more hardware incompatibilities. Sorry.

You will have to flash the orignal (!) firmware using the P6 jumper method as described in the wiki. If you don't have a backup of the firmware, you'll find someone to provide you with one. None of the builds we provide(d) will work and we have for obvious reasons not intention to change that. 

73
Danilo

db4ple, 2018-07-23:
BTW,

Aliexpress seller text is clear about this incompatiblity: https://www.aliexpress.com/item/10W-1-30M-MCHF-HF-QRP-Transceiver-SDR-Transceiver-AM-CW-LSB-USB-AM-FM-N/32819374713.html

KJ6ZH, 2018-07-23:
Richard,

There is a group https://groups.io/g/RS-918-HF-SDR-Transceiver where we are modifying the RS928Plus firmware to incorporate updates.  Also in the process of reducing images on FT8 reception.
The thread where this is being done is under 'Firmware v36 released'.

Also in the thread is the site to download the original manufacturer firmware.

73 KJ6ZH

KJ6ZH, 2018-07-23:
Anyone should be able to download the binaries from that thread.  Only need to join the site to post to it.

73 Chris KJ6ZH

db4ple, 2018-07-23:
Hi,
The GPLv3 requirement is to publish the source code !! not only a binary. 

Technically you (or whoever did the modifications) have to publish what is necessary to allow others to see how you modified the software and to make it possible for the users to change the source and build new binaries for themself.

KJ6ZH, 2018-07-23:
Will forward this requirement to that group.
Should not be a problem to include the source code as well.

KJ6ZH

closed 2018-11-04 by df8oe