df8oe/UHSDR #1560: can a second Firmware build be made that will be for freeDV and ssb only. All the other modes can be brought back by installing older Firmware

WM0I, 2018-08-15:
mchf 512K version. This is just a request for us who don't know how make a slimmed down version of the firmware to just do freeTV and ssb. Also freeTV takes care all the noise so the noise reduction and the notch filter is not needed. 

This should free up space for freeTV .
Please give this some consideration!

Best regards
WM0I   Richard

df8oe, 2018-08-15:
Hi Rich,

sorry. There is only a small amount of mcHFs out in the wilderness and starting such a huge refactoring will only ve a solution for a small time. Firmware is growing continuously and the border will be reached again at some time.

The only and clean solution is to swap the MCU, and if you do so you can choose one with 256K RAM and 2MB flash. You will be on the sunny sider forever.

We do not have time and manpower to release such "special configurated firmware" :/

But if you can fins a contributor who will do that: of course it is possible. But I think it is better to swap the MCU...

vy 73
Andreas

WM0I, 2018-08-15:
OK,OK I get it. Who can make the upgrade for me as I can not work on small parts any more. Can you give me people names and contact info.
Regards
WM0I

K4LXY, 2018-08-15:
Also, are there instructions somewhere about what to use and how to do it?
Tnx
73 Howard K4LXY

> 
> OK,OK I get it. Who can make the upgrade for me as I can not work on small parts any more. Can you give me people names and contact info.
> Regards
> WM0I

Howard Zehr
Photo galleries:  
www.howardzehr.com <http://www.howardzehr.com/>
https://www.facebook.com/HowardZehrPhoto

db4ple, 2018-08-15:
Use cheap hot air solder station
Heat processor pins carefully (train with old boards before, if you never did this before)
Remove processor once heated. Be carefull, do not pull to early.
Use flux and soldering iron and solder wick to clean pads
Use more flux and solder in new processor either with hot air or any other method which works with TFQP. See youtube. 
Check for shorts.
Check for shorts carefully.
Check for shorts even more carefully.

Then install bootloader and firmware according to first time instructions in wiki. 

You're done.  Or, then you should check for shorts, see above... ;-)

WM0I, 2018-08-15:
I guess I am out of luck because I am 73 yrs old and not able to work on SMD devices anymore. I will pay anyone with experience to make this change for me.

Regards
WM0I

WM0I, 2018-08-15:
Sorry, but Yahoo is not allowing sign-ins right now, I do have a account  But I can not login. Here again I keep asking for names and contact info, but I am not getting any.

df8oe, 2018-08-16:
You are here in the "firmware UHSDR development place". Your question is off-topic here so I think you will get no answer. Another place where you can ask for help is the [German hardware discussion group] (https://www.amateurfunk-sulingen.de/forum/index.php?board=15). There are already some US hams registered and so there is a chance for help.

closed 2018-08-23 by WM0I