df8oe/UHSDR #1561: rtty reversed ??

DG9BFC, 2018-08-15:
# UHSDR Firmare Defect Issue (Bug) Template 1.0

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Your Issue Data goes here:

title says all
when i try to receive some ham stations on 20m i have to switch to rtty L ... as far as i remember it was the other way around ...
we had to select rtty U in 30, 20m and up ,,, and select rtty L in 40 and 80m ... now i have to switch to L in 20m .... something mixed up??

Your firmware version:
Your bootloader version: 

### Hardware 
 * UI Board: mcHF 0.x / OVI40 1.x
 * RF Board: mcHF 0.x / OVI40 1.x

### Describe the issue:

Place tell us what does not work and how others can reproduce the issue (list related settings, steps to do, etc.):

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

You have audio related problems:
 * DEMODULATION MODE: LSB/...
 * Filter:  2K7 LPF
 * Noise blanker setting: 0
 * Noise reduction: ON/OFF

You have display (also fill if audio problem) related problems:
 * Display resolution: 320x240 / 480x320
 * Display connected via: Parallel / SPI
 * Touchscreen: Yes/No
 * Magnify: 1x / 32x
 * Spectrum: Waterfall/Scope/Dual
 
Other settings you consider as relevant:

 * ...

s53dz, 2018-08-16:
To my believe there is no real argument nowadays to go LSB and USB on different bands but legacy (... and an old ITU doc).

73 Bojan

df8oe, 2018-08-19:
I do not think this is a bug. It is related to personal working conditions and personal views...

closed 2018-11-04 by df8oe