df8oe/UHSDR #1562: A choice of 50mW TX out

s53dz, 2018-08-16:
A question for possible improvement.

Since mcHF is not equipped with a separate low power TX output I wonder if it would be helpful to have the lowest settable power 50mW instead of now 0.5W? It would certainly help me.

As I see it from the source it would require just one-line change in this file:
`drivers/ui/radio_management.c
`approximately at line 905

change
```
    case    PA_LEVEL_0_5W:
        pf_levelscale = 0.316;        // rescale for 10% of 5 watts (0.5 watts)
        break;
```
to
```
    case    PA_LEVEL_0_5W:
        pf_levelscale = 0.1;        // rescale for 1% of 5 watts (0.05 watts)
        break;
```
I suppose it does not meet any integer limits in calculation elsewhere.

73 Bojan S53DZ

s53dz, 2018-08-17:
Hi,

I can confirm that this simple change of TX output power from 0,5W to 0,05W works.
I have tested in TUNE and CW modes.
This would also call for a beauty change in display to show 50mW.

Test results of a particular setup just for information:
5.5W 
1.82W
0,9W
0,390W --> 0,040W

I have tested this on an OVI40 UI and mcHF RF.

73 Bojan

sp9bsl, 2018-08-17:
Hi Bojan,
I would advice to add another setting in power change ring, not simply change the current lowest 0.5W setting to another. Current low power is sometimes useable when testing unknown antenna or using amp which requires such level or just simply low distance test or... etc. Just my thought...

73 Slawek

s53dz, 2018-08-17:
Hi Slawek.
Thanks for your comment.
I have done that, cycling between FULL, 5W, 2W, 1W, 0.5W and 50mW. And a proper power display.
But you are right. This way the carrier always stays at the same level. Which is awful.

73 Bojan

s53dz, 2018-10-17:
Hi,

Just a short report to tell that in the mean time I have solved the issue locally by adding some logic to it.
Doing it quite old-fashion way. 

Activating PTT/ line I use a relay to switch between the RX and TX, where in the TX line I have added an 26 dB pi attenuator. This figure I need for the attenuation when using 0.5 W output power. This works well for me.

73 Bojan S53DZ

[image]