df8oe/UHSDR #1571: This is information only. No problem.

WM0I, 2018-08-30:
Here is some great information about firmware version 2.9.48. I have had problems with the PA form 2.9.01 not being able to set the bias correctly so that the power settings would work from the screens settings to reduce power output from 15 watts down to about 0 watts. I now have complete control of the power output so that I can the correct power level to drive 50 watt power amp.

FANTASTIC!  I don't know what you have done to fix the problem I had in the past versions. I looked at the code and did see where you are now controlling the PA bias from real time annlayas.  Now all I have to do is use the mic gain to reduce the power output.

Great job
WM0I Richard

df8oe, 2018-09-02:
There was nothing changed in that area...

WM0I, 2018-09-07:
There may have been nothing changed in the PA area as you say, but the software says there is a real time adjustment. Also if there was other areas changed it has helped in someway.

WM0I Richard

df8oe, 2018-09-07:
RTC adjustment is there since over a year. But it should only be displayed if you do have 1) SPI LCD and 2) RTC mod installed - or you do have OVI40 (there it is available always).
Otherwise it does not have any sense. And yes of course: each commit does improve something or is a bugfix. Not even influence for all hardware platforms.

closed 2018-10-18 by df8oe