df8oe/UHSDR #1577: Snap Function

regor60, 2018-10-08:
# UHSDR Firmare Defect Issue (Bug) Template 1.0

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Your Issue Data goes here:

Your firmware version: latest
Your bootloader version: 3.4

### Hardware RS-978
 * UI Board: mcHF 0.x / OVI40 1.x
 * RF Board: mcHF 0.x / OVI40 1.x

### Describe the issue: Snap function in CW often jumps 10s of Khz away from frequency

Place tell us what does not work and how others can reproduce the issue (list related settings, steps to do, etc.):

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

You have audio related problems:
 * DEMODULATION MODE: LSB/...
 * Filter:  2K7 LPF
 * Noise blanker setting: 0
 * Noise reduction: ON/OFF

You have display (also fill if audio problem) related problems:
 * Display resolution: 320x240 / 480x320
 * Display connected via: Parallel / SPI
 * Touchscreen: Yes/No
 * Magnify: 1x / 32x
 * Spectrum: Waterfall/Scope/Dual
 
Other settings you consider as relevant:

 * ...

DD4WH, 2018-10-08:
Hi
it would be good if you gave us some more information on when the issue occurs. It is very hard to interpret your problem.
In CW, snap mode is delicate because the carrier is not present 100% of the time. So if you have a broad filter and several carriers present, the snap mode always catches the strongest carrier which might not be the signal you want to SNAP.
There is another known issue: if you press SNAP in a mode which does not allow SNAP (USB/LSB), and then change mode (to a mode where SNAP is allowed), the SNAP algorithm changes frequency to the frequency SNAPPED in the old mode. I think this is the issue you experience, because you write "10s of kHz". That is not possible in the normal way SNAP is working since I rejected such large frequency changes in the software code. This happens only, if you press SNAP in a mode where you are not supposed to do so, and then change mode.
73 Frank DD4WH

regor60, 2018-10-08:
Thanks.

This has been occurring on 14.004 for example or close to the band edge
running CW and CW decode on with 300/700 BPF

When I have gotten close to zero beat by ear, pushing the snap button has
on one occasion sent the rig to 13.970 or so. That was the largest
excursion. When it happens it always drops, doesn't go up.

Hope this helps. Obviously if no one else reports this it's my local
problem, but a minor inconvenience 😀

Roger

df8oe, 2018-11-04:
Nobody reports same - I think it is a local problem and issue can be closed...

closed 2018-11-04 by regor60