df8oe/UHSDR #1579: Long USB descriptor strings prevent pulseaudio from creating audio output sink

miek, 2018-10-16:
# UHSDR Firmare Defect Issue (Bug) Template 1.0

Your firmware version: 2.9.52
Your bootloader version: 4.1.2

### Hardware 
 * UI Board: mcHF 0.6
 * RF Board: mcHF 0.6

### Hosts software
 * Ubuntu 18.04
 * Pulseaudio v11.1

### Describe the issue:

After connecting with USB, the mchf interface is recognised along with the serial port and audio device. The audio input device is added and works, but the audio sink does not appear.

I re-ran Pulseaudio with verbose logging enabled and found this error:

    D: [pulseaudio] source.c: Failed to register name alsa_output.usb-UHSDR_Open_Source_Community__based_on_STMicroelectronics_Driver_USB_Interface_mchf_00000000002A-02.analog-stereo.monitor.

Looking through the Pulseaudio source code showed that their maximum name length is 128 (shorter than the name above), so I patched the UHSDR source to shorten the manufacturer string and the audio sink now comes up correctly: https://github.com/miek/UHSDR/commit/a97ffcd04a72f722b29910a6005566de35ac97a8

df8oe, 2018-10-17:
You are right! We will release a fix soon. Actually I am working on some OVI40 hardware integration to firmware so I want to wait until I do have a break.

db4ple, 2018-10-17:
Thanks for finding and reporting this issue!

@miek: What of the string 
`alsa_output.usb-UHSDR_Open_Source_Community__based_on_STMicroelectronics_Driver_USB_Interface_mchf_00000000002A-02.analog-stereo.monitor`  
is actually counted in for the 128 bytes? 

In fact, our string is 100% legal USB and just 63 bytes long, so the real issue is on the Pulseaudio side since it will fail for all USB devices with too long (from Pulseaudio's perspective) manufacturer strings.

I am not saying we're not fixing it, but we should also notify the "owners" of the real issue since it may make the life more users easier. BTW, I was hit by this issue I think, since I could not get the mcHF to work with wsjt-x on Ubuntu 18.04.

df8oe, 2018-10-17:
This is not only Canonical specific - ALL pulseaudios of all distributions are working same way. I am using Siduction which is based on Debian experimental and LMDE3 which is based on Debian stable and both show same issue.

Complete string from beginning to end is counted - not only "our manufacturer string". When I use it on a MAC also strange things happen in logs. String is cut and overwrites following string begin. So I think we must reduce it. It is unclear how long it is allowed to be but I suppose not more than 25 characters.

miek, 2018-10-28:
Yep, it's fixed -thanks!

closed 2018-10-28 by miek