df8oe/UHSDR #1580: RTC Glitch

WD8BXS, 2018-10-18:
Your firmware version: 2.9.52
Your bootloader version: 4.1.2 

### Hardware 
 * UI Board: mcHF 6
 * RF Board: mcHF 6

### Describe the issue:

RTC mod finished, M1 and F3 buttons do not work

 * Display resolution:
 * Display connected via: SPI
 * Touchscreen: Yes
 

Thanks,
Chuck WD8BXS

df8oe, 2018-10-18:
RTC is not recognized correctly. This can be due to a hardware issue (32KHz oscillator not working, RTC battery not placed, soldering issue) or due to an "installation procedure error". It is recommended to activate RTC *exactly* the way it is described.

WD8BXS, 2018-10-18:
HA HA, I forgot to install the zero ohm jumpers.

Sorry de
Chuck WD8BXS

closed 2018-10-18 by df8oe