df8oe/UHSDR #1597: problems with updates  54 and 55

g7jiq, 2018-10-29:
Just wondering if anyone has had the same problem.
Mchf  mcu 512kb
Locks up on boot screen with 2.9.54 and 2.9.55. (see attached picture)
2.9.53 works fine.
[image]

db4ple, 2018-10-29:
Interesting. Honestly no idea what changes could make the 2.9.54/55 crash. I reviewed the changes between 2.9.53 and 2.9.55 and most of it is not related to the mcHF hardware or should have no impact.

Is the crash happening always at same time (i.e. the line "Firmware chan" is showing, possible a few more or les letters but roughly the same content).

If someone else has this, it is interesting to learn if this is a 512kb Flash or a larger machine.

df8oe, 2018-10-29:
I cannot reproduce the issue, because of I praid since 2015 "use MCUs with 1MB of flash" and this is probably the cause of the issue. All of my mcHFs do have >= 1MB of flash and do work flawlessly.

If you want to be future-proof it is a *very good idea* to swap MCU with a 2MB 429/439 MCU.

db4ple, 2018-10-29:
Hi, I tested the original downloads (both normal and small) and both work for me if not connected to an RF board. However, when connected to the RF board, both crash. Unfortunately, they don't crash if I connect an ST-Link debugger (and power it). There is something interesting happening here.

g7jiq, 2018-10-29:
Yes it crashes at the same place on both 54 and 55,
re-flashing it back to 2.9.53 boots up fine.
And swapping upto bigger cpu would be great, but I am worried about trashing it as its a great little radio and It would be a shame to loose it.
If there is no option, I would be happy to stick to the software ver that currently works.

Thanks for looking into it..

df8oe, 2018-10-30:
Another interesting information: what bootloaders do you use?

db4ple, 2018-10-30:
BTW, I think my problem (crash without ST-Link, normale boot with ST-Link plugged and powered) may be entirely unrelated, since I have the phenomenon also with the 2.9.53 firmware and it crashes much later (after showing the boot screen fully).  I'll have to investigate further.

g7jiq, 2018-10-30:
I am using bootloader 4.1.2,  
Button test works fine,

db4ple, 2018-10-30:
Thanks. Button test works. Good to know.

UR7FM, 2018-10-30:
Same problem with RS918.(512kB) re-flashing it back to 2.9.53 boots up fine. Bootloader also 4.1.2

df8oe, 2018-10-30:
Please test default settings boot.

db4ple, 2018-10-30:
Just press F1-F3-F5 during power-up and agree to  reset to default. Please note by cutting the power or pressing Band+  while pressing Power off, the reset will not be saved and you keep the existing configuration stored in non-volatile memory.

df8oe, 2018-10-31:
...it looks like this is same mystery as "16KHz issue" which was present some weeks ago. Most of users never have had seen it, some do have it everytime... Change of some lines in FW fixed it for all - but for the ones who never have had it chaging the lines does not have any influence on function...

UR7FM, 2018-10-31:
Just made test default settings boot.... Same problem:(

db4ple, 2018-10-31:
OK. Thanks. Is it crashing more less exactly like shown on the screenshot of @g7jiq ? 
What I could do is to build a few test builds where we try to isolate the area of issue. That would work like this:
I build something and attach it here. Then you test it and report back what you see  (i.e. if it crashes or not).

db4ple, 2018-10-31:
Just to report my findings: my problem with 2.9.x was related to my power supply current limit being set too low. With sufficiently high limit I can now boot the official 2.9.55 small and large without problems. 
So we are back to square one. I have an 192k MCU with 1MB flash, but that does not matter much, when flashed with a 2.9.55 small build.

db4ple, 2018-10-31:
I have more questions: Is the TRX connected to a PC via USB when booting? If so, try without connected USB cable.

df8oe, 2018-10-31:
Interesting idea. I have tried booting with connected USB cable to Linux and Windows PC: no luck - booting always flawlessly.

db4ple, 2018-10-31:
This was a shot in the dark. Anyway, the only relevant difference between 2.9.53 and 2.9.55 are the change to a different debug mode detection and some (unused) menu entries which should not make a difference. The HMC1023 itself is not active and some minor changes in the  CAT protocol. No (hardware) initialization code was changed. Since it crashes during printing on the screen, it is very,very likely that the crash comes from interrupts. However, we haven't  change any code there. And it must have some "external" dependency we don't know yet, otherwise a larger group of machines should be affected, most notably our test machines.

UR7FM, 2018-10-31:
Problem same as on the screenshot of @g7jiq
[image]
USB cabe was disconnected from PC , I updated FW via  USB memory stick allways., bootloader via dfu mode

df8oe, 2018-10-31:
Please test:
entering button test mode and then press&hold BAND- to reboot in powered mode.

db4ple, 2018-10-31:
Okay, good. This means it is a) exactly the same issue and b) it seems to be quite deterministic in its timing.
So, would you be ready to receive some test binaries?

db4ple, 2018-10-31:
@df8oe: The button test question was already answered, and yes, it works.  We go into button test earlier than printing this message.

UR7FM, 2018-10-31:
Yes , I am ready

db4ple, 2018-10-31:
Here we are. Flash it and tell us what happens. This software will not work correctly but may get you further in the boot process.

[fw-mchf.zip](https://github.com/df8oe/UHSDR/files/2534890/fw-mchf.zip)

UR7FM, 2018-10-31:
Now  show start up normally , but trx frozen no any button work

df8oe, 2018-10-31:
@db4ple : I know button test works. But maybe it is different behaviour to boot from "cold" or have a little delay by coming from input test and starting with BAND- Button - having all the time power supply for STM.

UR7FM, 2018-10-31:
No delay, same as older FW

db4ple, 2018-10-31:
Okay. As I said, this firmware was not supposed to work normally. Buttons not working is expected, since I disabled some critical interrupt to find out if this is the trouble maker. We can conclude that it at least has an influence. I will think and upload a new binary in a few minutes.

UR7FM, 2018-10-31:
ok

db4ple, 2018-10-31:
Here we are: Try this and see if it stops again at the same position.
[fw-mchf (2).zip](https://github.com/df8oe/UHSDR/files/2535050/fw-mchf.2.zip)

UR7FM, 2018-10-31:
Now work fb, no any stops! Where was problem?

db4ple, 2018-10-31:
Here comes the thing: I have no idea. The firmware you are running has been compiled with exactly the same source code as df8oe used to compile the 2.9.55 official release. Only difference is that I did this build using Windows and Eclipse. Even the compiler is probably the version. So this leaves us with an unidentified issue sleeping somewhere in the source code. Since I can also run the official build without problems (the one which did not work for you), we are not much wiser expect that the problem is with very high degree not in the actual changes we did but somewhere hidden. 
Please try to use the official binary 2.9.55 just one more time in order to verify that this one still crashes.
Thank you!

df8oe, 2018-10-31:
Iam actually using gcc 7-2018-q2-update as I did the last months. There is nothing during compilation that looks strange.

Strange was the "16KHz-issue". There are some parallels with this issue. I could not reproduce it here (except with scrambled sonfiguration) - but esmarties hav it all the time. Just like this issue. It is no proof, but I think there is a hardware dependency, too. Sadly an unidentified one vecause we could not take a look at a rig with issue using hardware debugging.

UR7FM, 2018-10-31:
With official release same problem
[image]

db4ple, 2018-10-31:
@df8oe: Could you do a "small/512k" build with https://github.com/df8oe/UHSDR/blob/active-devel/mchf-eclipse/drivers/audio/audio_driver.c#L737 commented out and upload here. 
@UR7FM: Please test this once uploaded...

db4ple, 2018-10-31:
Good that I checked: I was compiling with an older compiler. Now running the build for the same one df8oe used. Lets see what happens now on my side. And see if the build works for @UR7FM.

df8oe, 2018-10-31:
@db4ple can you build using Makefile on Windows? I wanted to build a "small build" using Eclipse but I failed. In spite of uncommenting #define IS_SMALL_BUILD the binary exceeds limit for small MCUs (its size is 468KB)

df8oe, 2018-10-31:
OK - I refreshed Eclipse and now it is a small build. Could you pse test Gennadiy? It ia small build using same code and same compiler but Eclipse wrapper.
[fw-mchf(3).zip](https://github.com/df8oe/UHSDR/files/2535418/fw-mchf.3.zip)

UR7FM, 2018-10-31:
Yes, it work well, alles in Ordnung:)

df8oe, 2018-10-31:
Different behaviour of Makefile and Eclipse builds using same compiler. Just a shot in the dark but I think it is a race condition which is uncovered by different optimization / different linking rank order. Danilo is right: there is something sleeping in the code we cannot identify without debugging at a real rig which shows the issue.

UR7FM, 2018-10-31:
Not working FW checksum: F0192336600201C108D1C7ECCEBD977B
working FW checksum: 055076D692123C1691204E72223893F6

df8oe, 2018-10-31:
My idea is that all those who do have 12 (or 24??) KHz issue *and* clean config does *not* work do have this issue, too. In my opinion it is the same, buried cause. To verify this every report is highly appreciated and welcome. Deiails as Danilo requested are mandatory.

ok1if, 2018-11-02:
Congratulation, version 2.9.59 doing with my clone RS918 withou problems (512kb flash). Good job
Milan OK1IF

ok8oi, 2018-11-02:
Exactly the same symptom! Stops at the very same place. However, everything was solved this morning with 2.9.59 in my RS-918. Congratulations from me as well!

g7jiq, 2018-11-02:
Just tried the new 2.9.59 software and all works fine, well done.
Thanks to all for all the effort involved.

db4ple, 2018-11-02:
Hi,
we are not done yet. We have no idea why the change we did, has this effect. It is a complete mystery to us, since the removed code is looking good. This should not cause any issue. So the problem may reappear at any time.

df8oe, 2018-11-02:
Many thanks for all your feedback! We have found a solution - but in my eyes it is only a workaround. We do not understand, why
- the issue pops up only at Makefile builds
- the issue is only present at a handful devices.

So we want to make further investigations. I attach a build "59" with some modifications. Would you please test if this works for you?

Many thanks - we cannot test it here at any device...
[fw-59s-1.zip](https://github.com/df8oe/UHSDR/files/2543616/fw-59s-1.zip)

df8oe, 2018-11-02:
@db4ple: I moved the code to the very beginning of main() - no changes to the code itself.... I am exited about the result...

ok8oi, 2018-11-02:
I just tried the zipped file a few minutes ago. Unfortunately it has the same behaviour. It is stacked. No control what so ever on it. Had to unplug the power to stop it...

ok8oi, 2018-11-02:
I went back to 2.9.59 from UHSDR Project Page and it works fine now... 
73 DE SV1OI/OK8OI

df8oe, 2018-11-03:
Many thanks for testing. We must investigate further - the cause of the crash is still unidentified. Please test attached binary, too.
[fw-mchf-test-60-1.zip](https://github.com/df8oe/UHSDR/files/2544726/fw-mchf-test-60-1.zip)

df8oe, 2018-11-03:
Another interesting question to all who do have the issue:
do you have populated oscillator or crystal (16MHz) for clock of STM?

df8oe, 2018-11-03:
regarding oscillator / crystal:
We are looking for something that is identical on all devices which do have the issue. In original mcHF an oscillator is published, so most devices ran using oscillator. But crystal is some penies cheaper than oscillator - so maybe clones do have crystals. If we know what is identical on all devices which do have the issue we can think further... All we are actually doing are shots in the dark.

g7jiq, 2018-11-03:
Ver 2.9.60 works fine and I have the oscillator block fitted.

df8oe, 2018-11-03:
@g7jiq : important information is that "official build" as also the binary from my post (fw-mchf-test-60-1.zip) is working.

g7jiq, 2018-11-03:
Just tried ver 60-1 from your post and all is well. (working fine)

df8oe, 2018-11-03:
Thanks for testing. That clarifies the issue is NOT located in the code which was temporarily removed - I just moved it to another place nearby and all is working. Still unidentified but one step further...

g7jiq, 2018-11-04:
Just tried 2.9.61 from the UHSDR Project Page and that works as well.

df8oe, 2018-11-04:
OK. Not understand the problem, but we are knowing that there is something sleeping. Thanks to all for testing! I think issue can be closed.

closed 2018-11-04 by g7jiq