df8oe/UHSDR #1610: 2.9.64 and 2.9.65 doesn't work

WM0I, 2018-11-14:
# UHSDR Firmware Defect Issue (Bug) Template 1.0

Please fill out the appropriate values.  Remove inapproriate/irrelevant values, but be prepare to provide more data. This template contains the most often asked questions.
We may adjust the template over time.

Please give as much information as necessary. At the same time, try to be concise. If you want to report something else than a defect, consider using a forum first. If you want report anything else but a defect remove the template if it does not make sense for your issue. Thank you!

## Issue Data goes here (please remove text above):

Your firmware version:
Your bootloader version: 

### Hardware 
 * UI Board: mcHF 0.x / OVI40 1.x
 * RF Board: mcHF 0.x / OVI40 1.x

### Describe the issue:

Place tell us what does not work and how others can reproduce the issue (list related settings, steps to do, etc.):

### Your relevant settings
Hint: most of the information can be seen on a screenshot of the main display.

You have audio related problems:
 * DEMODULATION MODE: LSB/...
 * Filter:  2K7 LPF
 * Noise blanker setting: 0
 * Noise reduction: ON/OFF

You have display (also fill if audio problem) related problems:
 * Display resolution: 320x240 / 480x320
 * Display connected via: Parallel / SPI
 * Touchscreen: Yes/No
 * Magnify: 1x / 32x
 * Spectrum: Waterfall/Scope/Dual
 
Other settings you consider as relevant:

 * ...

db4ple, 2018-11-14:
Hi,
could be explain, what you meant by "did not load", was the boot process not working? Or was it a problem on your side? 

Thanks!
Danilo

WM0I, 2018-11-15:
2.9.65 loaded but when ran the new firmware all I got on the screen was the upper right portion of a normal screen. Nothing worked. So I tried to reload 2.9.64 which worked before I loaded 2.9.65. Now when loading the 64 version I got an error 2 message at the bottom of the screen and it wouldn't run.  I am now at firmware version 2.9.63 and it works.

WM9I

WM0I, 2018-11-15:
Replace the USB stick and loaded 2.9.65 firmware. I got the same results but with No Error 2 message. When I tried to run the new firmware I get just the upper right hand one 4th of a normal screen. 2.9.64 firmware loads OK now.

WM0I

WM0I, 2018-11-15:
Boot screen seams to be working but the main screen is not. The best I can describe what I see is if you cut the main screen up into 4 sections 2 x 2 . Nothing works and I have to take power off to turn it off. 
WM0I

WM0I, 2018-11-15:
What I am seeing in system info is HY28B SP1 and 512k.  So I guess I need a newer UI board which is no longer available.
WM0I

db4ple, 2018-11-15:
Hi,
it is not HY28B SP1 but HY28B SPI -> aka serial display connection. You don't need a new UI board, we need to fix the software. I have an idea what the problem could be. And I also have the hardware to test this idea.  For now, simply stay at 2.9.64 until we release a fix.

df8oe, 2018-11-15:
I do have a mcHF with SPI-display and it works like a charm with 2.9.65 firmware. It uses STM32F429 MCU...

WM0I, 2018-11-15:
My CPU is 413h.     

WM0I

df8oe, 2018-11-15:
Hi Rich,

your MCU is F40x@512KB - the smallest one which is available / will work on mcHF. The number alone is not sufficient for description - the amount of flash must be added. Noone here has access to this MCU. 
But it is nearly impossible that amount of flash is the reason.

db4ple, 2018-11-15:
Hi,
@df8oe, Rich already gave us the 512k, but like you, I don't think it is the flash size. I have a very different hunch. See my next pull request.

df8oe, 2018-11-15:
Sorry - accidently closed.

df8oe, 2018-11-15:
I am exited if 2.9.66 has different behaviour at affected radio...

ok1if, 2018-11-15:
2.9.64 working normal, but 2.9.65 and 2.9.66 not (512 kB)

db4ple, 2018-11-15:
@ok1if:  What is happening? Same as @WM0I described, "broken" main screen?
Do you also have an SPI display or is it parallel?

db4ple, 2018-11-15:
@df8oe , should try to disabled your version check code? Just to see if this makes a difference?
I cannot reproduce the issue although my machine is very close to the one of @WM0I (STM32F407, 192k RAM, SPI display, major difference is 1024k vs. 512k flash).

df8oe, 2018-11-15:
This was my idea too: the problem is sleeping at an unidentified part. Unhappily I am nearly sure we will not find it until we do have access to a machine which shows the issue. I never have had seen such a machine.

df8oe, 2018-11-15:
Question to all: Does anyone have a MCU > 512KB flash which shows the issue?

UR7FM, 2018-11-15:
Same problem with 2.9.67 ::(

db4ple, 2018-11-15:
Could someone of you please do provide us with a dump of the settings using the uhsdr_tool, see 
https://github.com/df8oe/UHSDR/wiki/Backup-and-Restore#backup--restore-on-external-computer-since-2919 and attach the dump here. Maybe it helps us to reproduce the issue.

Thank you!

UR7FM, 2018-11-15:
Can't start uhsdr tool on my Windows7(32bit)...tried to run from a command prompt window but system not compatable...

db4ple, 2018-11-16:
@ok1if : Do you have the same screen issue, as UR7FM? If so, would you be willing to send your unit to us in Germany for testing? We are in need of a unit which does have the issue and on which we can use the real-time debugging using an ST-LINK probe. We'd cover the shipping cost, if that is an issue for you.

db4ple, 2018-11-16:
BTW, the configuration backup is still something I would like to try as well in order to possibly prevent the shipping of a mcHF to us.

WM0I, 2018-11-16:
Sending my unit tp you so that can resolve this problem is a great idea. My problem with it is that is the only radio I have. I am 74 years old I have very little money to by another one. Do you have unit to replace this one where we could just trade them?  Let me know if this is possible.

WM0I

la2fda, 2018-11-16:
Hi.
I have the same problems with version higher than 2.9.63.
Here is my uhsdr_config_json.
[uhsdr_config.json.txt](https://github.com/df8oe/UHSDR/files/2590003/uhsdr_config.json.txt)
Hope you can use it, and just ask if you want me to test anything else.

73 LA2FDA

la2fda, 2018-11-16:
![20181116_191025](https://user-images.githubusercontent.com/30241805/48639748-d1620280-
Hi again. 
I have a couple of mchf's
This one is v.6 from m0nka, but bought used from a ham in UK.

I uploaded images of all the system-info.

e9d4-11e8-9ac1-d6a7ba6a422a.jpg)
[image]
[image]
[image]
[image]

I also have self build v.4 from m0nka, with 1024k flash, but this one is broken at the moment, so I am not able to compare. Will try to fix it.....

You are doing a great job with this software ! 
73 LA2FDA

tomgit12, 2018-11-16:
Hi all,
have the same issue with a RS-918 clone.
Normally i don't bother you because i am using a clone.
I don't have the necessary solder skins. :-(

Here are my config, i also had the last issue with power up what was fixed from you with moving some code lines.

[image]

[image]

[uhsdr_config.json.txt](https://github.com/df8oe/UHSDR/files/2590465/uhsdr_config.json.txt)

Thanks for your work, you are doing a great work with developing the code!
Can provide my unit if it is necessary.

db4ple, 2018-11-16:
Thanks, guys!
So far we have only 512k machines showing the problem. But it affects parallel and serial displays alike.
My machine is 1024k but I still cannot see how this could make a difference.

tomgit12, 2018-11-16:
Small idea from my side, but i am not a programmer here so can not say if it could be a problem.:
There is a known prefetch issue with this cpu.
[link](http://www.onarm.com/forum/62262/)

db4ple, 2018-11-16:
Hi Tom,
you might have found something. I will build some test binaries and attach them here. Let us see if it helps.

tomgit12, 2018-11-16:
Glad to hear, i can test if you want.
[Some ](http://ds.arm.com/media/resources/db/chip/st/stm32f417/DM00037591.pdf) more informations for you maybe.....

db4ple, 2018-11-16:
Thanks Tom, I was already looking at this document when your message arrived to waste some time while the compiler is working :-)

db4ple, 2018-11-16:
This one should crash. If it does, please tell me which led is still on (red and/or green). 
[fw-mchf.zip](https://github.com/df8oe/UHSDR/files/2590561/fw-mchf.zip)

db4ple, 2018-11-16:
Ok, here we go:
[fw-mchf.zip](https://github.com/df8oe/UHSDR/files/2590606/fw-mchf.zip)

la2fda, 2018-11-16:
> Ok, here we go:
> [fw-mchf.zip](https://github.com/df8oe/UHSDR/files/2590606/fw-mchf.zip)

Display and both leds are dark all time with this one.
..... Seems like I have trouble loading the old one now.......

tomgit12, 2018-11-16:
> This one should crash. If it does, please tell me which led is still on (red and/or green).
> [fw-mchf.zip](https://github.com/df8oe/UHSDR/files/2590561/fw-mchf.zip)

nothing happens, no screen no light. only display flashes 1time fast

tomgit12, 2018-11-16:
> > Ok, here we go:
> > [fw-mchf.zip](https://github.com/df8oe/UHSDR/files/2590606/fw-mchf.zip)
> 
> Display and both leds are dark all time with this one.
> ..... Seems like I have trouble loading the old one now.......

Bootloader still available if you press BAND DOWN & POWER for flashing with usb. Did work for me to get back to the running version

la2fda, 2018-11-16:
> Bootloader still available if you press BAND DOWN & POWER for flashing with usb. Did work for me to get back to the running version

Tnx Tom. I Have to find a USB stick., - Long time since i used USB for this. :-)

db4ple, 2018-11-16:
Sorry, my fault. Wait until the next one. Power and Band+ should get you DFU, at least it did for me.

db4ple, 2018-11-16:
This one please:
[fw-mchf.zip](https://github.com/df8oe/UHSDR/files/2590657/fw-mchf.zip)

tomgit12, 2018-11-16:
Running here! 2.9.65

db4ple, 2018-11-16:
Hi, 
now comes the real interesting one:
[fw-mchf-prefetch.zip](https://github.com/df8oe/UHSDR/files/2590694/fw-mchf-prefetch.zip)

tomgit12, 2018-11-16:
> Hi,
> now comes the real interesting one:
> [fw-mchf-prefetch.zip](https://github.com/df8oe/UHSDR/files/2590694/fw-mchf-prefetch.zip)

Also running ? Buildtime 20:47 ....

db4ple, 2018-11-16:
I suspected that this is also running. I will have to do a  small trick now. Will take a few minutes.

ok1if, 2018-11-16:
I have the same configuration on RS918 like tomgit12 and with 2.9.65 build 20:47:10 working normal
Milan

la2fda, 2018-11-16:
> Sorry, my fault. Wait until the next one. Power and Band+ should get you DFU, at least it did for me.

No Problem. I am back in business :-)

ok1if, 2018-11-16:
Of course 512kB

ok1if, 2018-11-16:
[image]

ok1if, 2018-11-16:
[image]

ok1if, 2018-11-16:
[image]

db4ple, 2018-11-16:
Hi,
this is bin only, but Tom seems to be able to use it:
[fw-mchf.zip](https://github.com/df8oe/UHSDR/files/2590754/fw-mchf.zip)

db4ple, 2018-11-16:
Okay guys,
I will also call it day now. 
We made some progress, but I still think I need one of the problematic units on my desk. As you can see from my call sign, I am in Germany and a shipping from Europe would be the quickest way to get this done.

WM0I, 2018-11-16:
I take it you and the others have a good start at solving this problem and don't need my small unit for trouble shooting.. Good luck, I will wait fir the fix.

WM0I

ok1if, 2018-11-16:
[image]

ok1if, 2018-11-16:
Sorry dont working

ok1if, 2018-11-16:
The last function version

[image]

df8oe, 2018-11-16:
Very sad that "after all these years" of my praying "do not use 512KB MCUs" there are build devices using this MCU... I expected only very old radios which uses this MCU are alive - but all of you do own newer radios :/ But I am nearly sure if we have acess to one of this we (especially db4ple) can fix it. This will give you some more time until again something must be stripped off so that firmware will fit ;) @WM0I :it would be a very wise decision to ship the radio e.g. to Chuck to swap the MCU...

WM0I, 2018-11-16:
OK, how do I contact chuck? Thanks.

WM0I

df8oe, 2018-11-17:
Look for WD8BXS in our German discussion board and send him a pm. I am nearly sure he can and will help you. This would be the best and a future proof solution. If you have fitted 427/429/439 with 2MB flash and 256K RAM you do have bleeding edge for mcHF -not possible to fit a more powerful MCU to mcHF (or clones).

db4ple, 2018-11-17:
Hi Guys,
we just released 2.9.68. Please test and report.

tomgit12, 2018-11-17:
> Hi Guys,
> we just released 2.9.68. Please test and report.

No, same issue again. Screen only partial ....

[64 more replies]

commit 86154064 (2018-11-15): Merge pull request #1611 from db4ple/ovi40h7

Attempt to fix #1610: Delay reaction to external interrupts until full init
commit dc4edbbb (2018-11-15): Attempt to fix #1610: Delay reaction to external interrupts until full init
We do not process external interrupts before everything is initialized and ready.
This affects only PTT and CW paddle input.
closed 2018-11-24 by WM0I