df8oe/UHSDR #1613: Spectrum Magnify bug

ac0ax, 2018-11-18:
Your firmware version: 2.9.68
Your bootloader version: 4.0.0

### Hardware 
 * UI Board: mcHF 0.4
 * RF Board: mcHF 0.4

### Describe the issue:

If I go toTouchsceen via menu and select  Spectrum Magnify and try to change it the radio locks up and makes a tone.

db4ple, 2018-11-18:
Hi,
cannot reproduce. Please be more specific (which mode is running, from which zoom level to which), preferably attach your config here, see https://github.com/df8oe/UHSDR/wiki/Backup-and-Restore#backup--restore-on-external-computer-since-2919

df8oe, 2018-11-18:
Not reproduceable here, too.

df8oe, 2018-11-18:
Missing info: CPU type / flash.

ac0ax, 2018-11-23:
sorry cpu is 413 512k

db4ple, 2018-11-23:
Hi, 
maybe you were also affected by the Bootloader Issue #1610, please update your bootloader and test again. But I recommend to wait for the bootloader 5.0.1, which is fixed AND boots as fast as the old one and can do warm reboot (5.0.0 is fixed but boots slower and can't do reboots properly). Just for testing 5.0.0 is okay but you should really go for 5.0.1 once Andreas releases this one. 

73
Danilo

ac0ax, 2018-11-23:
i just updated the bootloader to 5.0.0 and now the issue is gone,

closed 2018-11-23 by ac0ax