df8oe/UHSDR #1623: Makefile Commit 20/11 breaks building on OSX

rensd, 2018-11-25:
## Issue Data goes here (please remove text above):

Your firmware version: 2.9.72 -> 2.9.74
Your bootloader version: 5.0.2
(Please note, we will generally not accept issue reports if your bootloader is not at least 5.0.1)

### Hardware 
 * UI Board: mcHF 0.6
 * RF Board: mcHF 0.6

### Describe the issue:

Compile fails on OSX 10.11.6 after update Makefile 20/11/2018, the compiler used is
- gcc-arm-none-eabi-4_9-2015q3
- the file firmware_obj_list.lst is not created
- F4 CPU
- Reverting commit this fixes the prolbem

db4ple, 2018-11-25:
Hi,

thanks for reporting. If the commit fails, your make is too old. The make function $(file ...) we are using is there since gnu make 4.0 from around 2013. 
So upgrading make on your end would be the easiest fix.

If we simply revert, we would loose the just gained ability to build on Windows systems using make.

There are workarounds possible, I think but these require some thinking and experimentation.

Just a side note: Your compiler is very, very old. If possible, upgrade to something newer. We currently build using gcc 7.3.3 (However, your gcc version is not related to this issue)

m-chichikalov, 2018-11-26:
Hi all, I would like to rewrite make to eliminate such this limitation as long as guys with OSX 10.11.6 will support me... But I think it's going to be wasting time as Make below 4.0 is really old...

rensd, 2018-11-26:
Hi,

thanks for the feedback, will update make my version GNU Make 3.81.
The reason for gcc 4.9 is that is a full build including FreeDV still will fit in a
512k F4 cpu , now 2.9.74 @ 418k. no other code tweaks needed.

df8oe, 2018-11-26:
That's correct. But sometimes we are very near to "does not work" regarding horse power and optimization. Optimization is not only done regarding size of binary but regarding speed, too. Step to newer gcc has brought speedup - I have tested. So using old gcc is "not officially supported". We do not test speed limits...

rensd, 2018-11-26:
Thanks for the clarifacation on speed optimizations, not using mcHF to the limit right now.
If I run into problems will first re-compile to F4-512k before filing any bugs.
Update **make** fixed the Makefile problem.

73's Rens

closed 2018-11-26 by rensd